nightscout/AndroidAPS #2: [2.7.0] [graphical/UI] Scaling of additional graph may need improvement

goomStar, 2020-09-24:
Reposting from https://github.com/MilosKozak/AndroidAPS/issues/2980#issue-708285027

btw, master build via github/nightscout appears as '2.7.0-dev' in about-section

The scaling of fresh additional graphs seem to depend on the first data source selected. For full graph functionality a pre-check before drawing would be required, to find the optimal scaling.

Example/Reproduction:
Create an empty graph and use COB and DEV as data. As COB is the first source, the graph will use a fixed scaling with positive numbers only, even though DEV would need negative values.
It also affects the deviation-slope in negative area. They always work stand-alone, but in combination with COB as first source, the graph is not optimized.
When no COBs are found to be displayed for the chosen timeframe, the graph stays blank completely (like in the picture).

[image]

dlvoy, 2020-09-24:
`-dev` suffix is only at *tag* 2.7.0, master branch (one commit later) have it fixed

goomStar, 2020-09-25:
Not fixable at current state.

closed 2020-09-25 by goomStar