nightscout/AndroidAPS #6: [2.7] Engineering mode enabled per default

ArthurusDent, 2020-09-25:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Engineering mode is enabled per default. Right now I can't disable it. I removed the file and rebooted but engineering mode is still enabled as shown in About.

Phone: Oneplus 3T (Android 9)
Pump: Medtronic 554
xDrip+ G6

Logs:
[AndroidAPS.zip](https://github.com/nightscout/AndroidAPS/files/5283082/AndroidAPS.zip)

goomStar, 2020-09-25:
Can confirm same behaviour.

veterlch, 2020-09-28:
In my case, not using git check-out but using project from version control, I could build 2.7 master version.
(2.7.0-dev was made when I used git checkout).

I hope it will work for you.

PR #12 (2020-09-29): [FIX] Make enginering mode enabled only when turned on
Fixes #6, bug that made engineering mode enabled for all on non-development branch or not-full flavours
After fix isEngineeringMode() is back simple getter to private property that checked if engineering mode was enabled
closed 2020-09-29 by ArthurusDent