nightscout/AndroidAPS #9: [2.7] no "import settings" and scrolling possible because of small display size

MarieT1D, 2020-09-27:
Tested with a "Jelly Pro" from Unihertz
If I want to import settings, I only see this and it isn't possible to scroll and click on "ok":
[image]
after minimizing the size of the file direction:
[image]
it is still not possible because of the next window (I can't scroll and click on "import")
[image]

osodebailar, 2020-09-27:
Maybe the same setting as in atom ca solve the problem.
Set the fontsize in settings of jelly to 'small'.
If this does not help, you can do a fast workaround until as solution is implemented:
Search the text 'Du wirst nach dem Master-Passwort gefragt. .......' and delete the second sentence 'Mit diesem werden die importierten Einstellungen entschlüsselt' in strings.xml.
Commit the the file and rebuild master

MarieT1D, 2020-09-27:
Thanks for the hint, but unfortunately it didn't work.
As in the third picture, there are too many listings, so shortening the phrases is not enough and changing the font size doesn't apply for me with this :/

osodebailar, 2020-09-27:
Than the Next workaround , Remove the unneeded Backup Filed from AAps/preferences Directory

MarieT1D, 2020-09-27:
There is only one at AAPS/preferences (which I need to import) or which to you mean?

osodebailar, 2020-09-27:
Ja Marie , alle Backups aus dem Ordner löschen die Du nicht brauchst. dann ist die Liste kurz und es sollte passen

MarieT1D, 2020-09-27:
The list I mean contains the information of the back-up (Name, Date, phone used for the back-up, AAPS Version used for the backup, etc.) (3rd picture)

osodebailar, 2020-09-27:
Ach das meinst Du. Sry. das war für mich keine Liste.
Du kannst alternativ testen den Title 'Einstellungen Importieren' zu kürzen. der Header verbraucht viel Platz. 
Für eine endgültige Lösung muss das Layout überarbeitet werden.

MarieT1D, 2020-09-27:
Already tried that, too, I changed all string variablies I could which have line breaks. The only thing I could still try is changing the size of the header.

osodebailar, 2020-09-27:
Ja da muss man sich Layoutmässig was einfallen lassen.
Wenn es nur um den Import geht , dann öffne die DateI : PrefImportSummaryDialog.kt
und kommentiere die folgenden 2  Zeilen aus 
Zeile 46 
//  val table = (innerLayout.findViewById<View>(R.id.summary_table) as TableLayout)
und Zeile 79 
 // table.addView(rowLayout, idx++)

dann siehst Du keine Zusammnfassung mehr , kannst aber imortieren

dlvoy, 2020-09-27:
Andreas, weil ich ein bisschen verstehe, nicht jede hier ist Deutschsprachige ;)

It is obviously error with summary dialog, which does not allow scrolling on ultra-small screens. I will fix that.
In the meantime - hack mentioned above will allow to import prefs.

osodebailar, 2020-09-27:
Dafuer gibt es Google . Da geht auch English zu deutsch und und umgekehrt . 
Aber ich hatte eh Marie direkt geantwortet .

swissalpine, 2020-09-27:
Yes, I deleted the German language files because the texts are too long and cumbersome for me. But it also seems to me that my font size is smaller, so I have no problems at all on the Atom. Maybe you should check the font size in the Atom settings (Einstellungen > Display > Erweitert> Schriftgröße).
[image]

MarieT1D, 2020-09-27:
> Ja da muss man sich Layoutmässig was einfallen lassen.
> Wenn es nur um den Import geht , dann öffne die DateI : PrefImportSummaryDialog.kt
> und kommentiere die folgenden 2 Zeilen aus
> Zeile 46
> // val table = (innerLayout.findViewById(R.id.summary_table) as TableLayout)
> und Zeile 79
> // table.addView(rowLayout, idx++)
> 
> dann siehst Du keine Zusammnfassung mehr , kannst aber imortieren

Worked, thanks!

dlvoy, 2020-09-29:
After first attempts, seems like AlertDialog.builder inner layout cannot just simple be wrapped with ScrollView. 
Probably whole dialog need refactoring from system-type Dialog to custom Activity, but this will be more complex fix.

osodebailar, 2020-09-29:
@dlvoy i have a solution that work. in the next two days i make a pull request for a review.

dlvoy, 2020-09-29:
Ha! I also worked on it in parallel :D here my attempt:
https://github.com/dlvoy/AndroidAPS/tree/hotfix/import-prefs-layout-uniherz

Do you have device to check it? I only had tested that on Emu.
Please feel free to choose which of those two fixes is better, I can review / emu test yours as well!

osodebailar, 2020-09-29:
I will have a look tomorrow on your solution 
I implemented my solution in my GUI Branch , if you want to have a look.
One of the solution for very small displays is here to set the title as marquee element , and only show the file name , not the path. The path is not interesting for a normal user.
The 3rd solution is in the end when the table in the inner element is shown. I set a nested scroll around .
You can test it here in my “new_gui_to_dev’ Branch 
https://github.com/osodebailar/AndroidAPS

MarieT1D, 2020-09-30:
I will test them, thanks for your work

osodebailar, 2020-09-30:
@MarieT1D you can test my  solution, i have a pull request too of you want.

dlvoy, 2020-10-02:
Fixed by @osodebailar with https://github.com/nightscout/AndroidAPS/pull/15 - so it can be re-tested from `dev` branch.
@MarieT1D  If test on device confirms it is solved - please close issue ;)

swissalpine, 2020-10-02:
I can confirm: The fix works on Unihertz Atom.

MarieT1D, 2020-10-02:
Yesss, it works, thanks so much for you work and sorry for the delay, I will close the issue

closed 2020-10-02 by MarieT1D