nightscout/AndroidAPS #18: [2.7] Feature Request: Add Battery Age, Canula Age and Remaining Glucose to Automation

Arne1346, 2020-09-30:
Hi,

I would like to use AAPS Automation do show me a notification when 

- Battery age is older than n days
- Canula age is older than n days
- Remaining glucose in ampulla is less than n IE

and this info I'd like to receive during daytime only.

Automation would be a perfect tool if the three triggers were available.

Could you add them?

thank you,
Arne

reijoahola-nightscout, 2020-10-01:
I would like to add also Battery level into this list. However, I must agree that all these can also looked after quite well with Status lights in the main screen.

Arne1346, 2020-10-01:
In principle true, compared to the previous AAPS versions the main menu display in 2.7 is a big step ahead, but still can be overlooked quite easily. My preference would be a notification you actively need to confirm.

MilosKozak, 2020-10-13:
setting tresholds and use colors is enough from my point of view
if you you watch the screen you'll see it. if not, notification doesn't help much

closed 2020-10-13 by MilosKozak