nightscout/AndroidAPS #19: [2.7] Minor bug: "Pump Site Change" always logged with "Insulin Cartridge Change"

Arne1346, 2020-09-30:
Hi,

sometimes I change the insulin cartridge, but not the canula.

When logging the event using the Prime/Fill action, I tick "Insulin Cartridge Change" only (do NOT tick "Pump Site Change"), but nevertheless also a "Pump Site Change" is logged.

To my mind this is not correct and a "Insulin Cartridge Change" should be logged only.

could you have a look?

thank you,
Arne

Philoul, 2020-09-30:
Do you have an Insight pump?
There is a little bug in Insight driver, when you change cartridge, a site change is recorded in AAPS even if you don't prime canula in pump.
A workaround is to disable "Log site change" in Insight preference, but it's necessary after that to always manualy record site change in AAPS even if done directly in pump...

reijoahola-nightscout, 2020-10-01:
Philoul, I have Insight v2 but when I change cartridge, AAPS automatically records "Insulin Cartridge Change" but not "Pump site Change". In Insight Preferences I have disabled "Log tube changes" but not "Log site changes". However, I never prime canulla in pump but use AAPS button.

NickSills1, 2020-10-01:
Similar here:

https://github.com/MilosKozak/AndroidAPS/issues/2972

Arne1346, 2020-10-01:
Hi,

No. Dana RS

/Arne

NickSills1, 2020-10-01:
Hi 

> No. Dana RS

That's why I said 'similar' not 'same'.  If it's pump specific then it's 2 problems rather than one.

Philoul, 2020-10-01:
@Arne1346 Can you please update issue title with "Dana RS pump" to avoid confusion.
I'll test again with my Insight pump and open a specific issue if I still have this bug.

Arne1346, 2020-10-02:
@Philoul : of course - done :-)

closed 2020-10-26 by MilosKozak