nightscout/AndroidAPS #21: NSClient disconnection on WLAN

Duester64, 2020-10-01:
AAPS 2.7
NS Client stops sporadically, if Wlan is on, saying destroy and not allowd until Wlan is switched off. then it reconnects. The Wlan signal is always stable and strong. This occured since update from 2614 to 2700.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Arne1346, 2020-10-01:
Not sure whether AAPS is the source of this issue.
Received an email from Martin who manages 10be.de with the info about recent issues. Martin was explaining that with the recent NS update to 14.x a huge number of additional connections to the Mongo DB are kept open which cause some pain on the 10be.de side. He is actively working on that.

Maybe this is also the origin of your reported issue - which for me (using 10be.de also) is simliar.

Duester64, 2020-10-01:
Have changed the issue, Seems to happen on a repeater change. I tried it just a minute ago.

MilosKozak, 2020-10-13:
not allowed means you have some restrictions is nsclient settings. check it
the info about current connectivity, charging etc is received from android.

i don't believe there is anything we can improve because we have to alwayse rely on android

closed 2020-10-13 by MilosKozak