nightscout/AndroidAPS #26: Pumpe ICT - lokales Profil mit falschen Kombinationen

franzec, 2020-10-03:
Alle Versionen
Wird ICT gewählt und anschließend das Profil editiert, so ist dort eine Basalrate zwingend anzugeben.
Da bei ICT die Basalrate durchLangzeitinsulin erfolgt, dessen Wirkkurve nicht dokumentiert ist, ist die Angabe U/h ohne Dokumentation nicht sinnvoll. Es ist unklar, wie AAPS dieses Langzeitinsulin einrechnet.
Wird es nicht im Bolusrechner berücksichtigt, muss auch der Wert 0.0 zugelassen werden.
Wenn ICT künftig erhalten bleibt, muss zumindest die Doc angepasst werden.

Philoul, 2020-10-11:
Can you please write in english?

maheini35, 2020-10-11:
I'll translate:
Pump ICT - local profile with wrong combinations
All versions
If ICT is selected and afterwards the profile is edited, then you'll need to type in the basal rate (obligatory).
But on ICT you'll take long acting insulin, whose action curve isn't documented. Because of this, the input as u/h doesn't make any sense.
It's unclear, how AndroidAPS is calculating those inputs. If it's not calculated in the wizard, then a basal of 0 should be accepted.
If ICT will remain, there should be at least a adjustment in the docs.

Philoul, 2020-10-12:
Thank's for translation (ICT = MDI in english)
For long acting insulin, I suppose the best solution is to consider a constant basal rates with 1/24 the total amount of daily long acting insulin.
For example if you have 20 U of very long Acting insulin every day (i.e. Lantus or Toujeo in France), I think you can enter just one basal rate value of 20/24 = 0.83 Unit / Hour
=> It's approximate and certainly less precise than with a pump, but that's how my basal rate was initialized when I switched to the pump.
I agree for doc if not documented, we will propose an update (Doc is also available in github...)

MilosKozak, 2020-10-12:
basal 0 is impossible because of many internal things. thus there must be some. but because you cannot change TBR it doesn't matter because basal IOB is always zero

rICTx-T1D, 2020-10-17:
Hi folks,
because I'm not sure that the answer reach @franzec, I will be translate to German for him:

>>>>
Der einfachste Weg (bei ultra-lang-wirkenden Insulin) ist es die gespritzte Menge an Basalinsulin durch 24h zu teilen und gleichmäßig im Profil auf die 24h aufzuteilen. Wenn du z.B. 20 IE von Lantus/Toujeo spritzt, ist das eine Basalrate von 0,83h je Stunde.

Philoul bestätigt, das die Dokumentation diesbezüglich aktualisiert werden könnte und diese ebenfalls im github ist. (Hinweis von mir: du kannst die Doku als selbst ergänzen, wobei die English-Version vorrangig ist, bevor die nationalen Übersetzungen kommen.)

MilosKozak sagt, eine Basalrate von 0 ist nicht möglich, da diese an vielen Stellen berücksichtigt wird. Da du mit der ICT die temporäre Basalrate nicht wie bei einer Pumpentherapie verändern kannst, ist das Basal-IOB immer 0.

So ich hoffe ich konnte dir etwas helfen, bitte schließ den Fall, wenn keine weiteren Fragen existieren.
<<<<

I hope I can help everyone to close this issue as soon as possible.
@Philoul: The insulin Toujeo is available in Germany too, so I use it since aprox. 2 years.

Bye
rICTx-T1D

franzec, 2020-10-17:
Thank you, I'll close the issue
Bye

franzec, 2020-10-17:
I'll try to change the doc.

closed 2020-10-17 by franzec