nightscout/AndroidAPS #27: [2.7] App stops suddenly without error

Duester64, 2020-10-04:
AAPS 2.7.0

App stops suddenly for some hours during night.

Display opens, but without glucose values during stop
No clucose values in lock sccreen, will be filled when algo resuming
algorythm stops, only basal running, dark blue color on insulin curve
no error message at all
App oens  when button on phone is pressed, then algo resuming
patched dexcom running without problems

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/5322898/AndroidAPS.log)

[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

pjdjesus, 2020-10-11:
App did crash without apparent reason. The app is exempted from battery optimisation and it is allowed to run in background Log uploaded. 
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/5362053/AndroidAPS.log)

pjdjesus, 2020-10-12:
The app stopped again. Exactly 24h after the last crash.

pjdjesus, 2020-10-13:
Probably is just me...The app stopped again at the same time 22:51. I've checked and does not seem been killed by any app on the phone.  
[AndroidAPS._2020-10-13_22-51-11_.0.zip](https://github.com/nightscout/AndroidAPS/files/5374129/AndroidAPS._2020-10-13_22-51-11_.0.zip)

cascer1, 2020-10-14:
I am also experiencing this issue. Every now and then I notice the app is stopped. Then when I start it again, it has to re-initialize (connect to pump etc) as if I rebooted the phone.

Using OnePlus 7 Pro running OxygenOS 9.

pjdjesus, 2020-10-18:
A cache reset followed by a clean installation solved my issue.

Satai1974, 2020-10-18:
I'm also experiencing this problem. As cascer1 described it looks like after a reboot, which is definitely not the case.
Using a new Nokia 5 with Android 9, only for looping, no other apps installed

cascer1, 2020-10-21:
As mentioned by pjdjesus, clearing cache and reinstalling the app seems to have solved the issue for me. I guess it's somehow related to upgrading from 2.6 to 2.7

Satai1974, 2020-10-21:
I think I've fixed my problem. I deactivated google play protect on my mobile and since then all apps run smooth and without any stop or hang or something like that. I have my nokia only for diabetic things, no other apps installed, so I don't think deactivating play protect is problematic. On other mobiles in every day usage this may be different? Anyway, my problems are solved so far.

MilosKozak, 2020-10-25:
@Duester64  does this work for you?

Duester64, 2020-10-26:
I have put off the automated optimization under device care and this worked for me. It is in the small three dot menue above on the S10.

closed 2020-10-26 by MilosKozak