nightscout/AndroidAPS #28: Dex App is losing connection to aaps 2.7.0 several times a day

FreDiabetics, 2020-10-04:
My Dex App is losing connection to aaps 2.7.0 several times a day. 

The only way to repair, is to toggle bluetooth off/on then aaps is recieving values again immediately.
Even the BT Watchdog doesn't seem to help here.

I'm using patched Dex G6 app with 81xx S/N Transmitters on a Samsung Galaxy S20 5G, Android 10

Here is a screen record of the issue. I'm using a different aaps repo with a new GUI but its at 2.7.0 master version. This issue occurs in the original 2.7.0 master too. Its not depeneding on the version I'm using.
https://streamable.com/3j3fdz

[image]

sfriedriszik, 2020-10-09:
I have the same problem.  Patched Dex App, AAPS 2.7.0 original branch.

rICTx-T1D, 2020-10-09:
I have the same problem, too. Patched Dex App "com.dexcom.g6.region3.mmol-patched.apk", AAPS dev branch from 2020-10-09, Samsung Galaxy S9, Android 10.

reijoahola-nightscout, 2020-10-10:
The same problem was in my system yesterday. Started at 16:00 (+3h GMT) and continued almost all the time about 6 hours. BT off/on was the only way to have ago time to start from 0 min but after next 5 min it again counted into 10 min and skipped back into 5 min. My system is patched Dexcom app g6.region1.mmol (version 1.5.1.5). AAPS 2.7.0 main branch. "NS upload only (disabled sync)" = ON. NS not migrated yet. The problem has not occurred before yesterday even I have used 2.7.0 now about two weeks. There were also several other AAPS users yesterday in Fb discussion who had the same problem and the problem started yesterday afternoon.

FreDiabetics, 2020-10-10:
@MilosKozak 

Any thoughts on this?

CaroGo, 2020-10-10:
Seems to be quite a problem for a lot of people! Missing values are really annoying and a quite important issue!

MilosKozak, 2020-10-11:
no idea

FreDiabetics, 2020-10-12:
@MilosKozak 

really? This is an issue which harms many people in using their loop effectivly..... 

Actual bg-readings are essential for a sensible bg managmanet. Please have a look into this.

sfriedriszik, 2020-10-13:
@MilosKozak 
since yesterday I turned off the NSClient in AAPS.  since then I have no more problems with missing blood glucose values.  is it possible that it is due to the NSClient?

MilosKozak, 2020-10-15:
if nsclient is crashing teoretically yes. but i don't know about it

MilosKozak, 2020-11-22:
Closed by inactivity

closed 2020-11-22 by MilosKozak