nightscout/AndroidAPS #35: [2.7.0]Minor bug: openaps SMB prediction not showing while CGM reading keep unchanged ~45m

maheini35, 2020-10-12:
See pictures. It only occures with the smb algorythm and only when my glucose levels are keeping strait. Since installation of v2.7.0 last afternoon, I had it twice (yesterday noon + this morning), always with the same symptoms and "error".
It's confusing, because you can't see any prediction and therefore don't know if there happened a big issue.
At the time the error ocurred, I was using temp target, but I mostly use it, so - I guess it won't affect the bug.

Predictions where always enabled, CGM readings where correctly :) - I tried the AMA algorythm and everything was fine. Also with past versions of SMB algorythm.

I'm using:
AndroidAPS 2.7.0
Build: bc43a926f-2020.10.11-22:16
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout-Version: 14.0.6
Developer-Mode: activated
Accucheck Insight firmware 3.x
Xdrip as BG source -> unit mmol/l
Libre 2 with patched app
Huawei Mate 10 pro, Android v. 10

//AndroidAPS screen
[image]

[AndroidAPS_LOG_1602492969986.log.zip](https://github.com/nightscout/AndroidAPS/files/5364142/AndroidAPS_LOG_1602492969986.log.zip)

[Scrollshot SBM calculation sreen.zip](https://github.com/nightscout/AndroidAPS/files/5364209/Scrollshot.SBM.calculation.sreen.zip)

ArthurusDent, 2020-10-12:
This is not a bug. You can read about it here:
https://github.com/MilosKozak/AndroidAPS/issues/2868#issuecomment-678638468
https://github.com/MilosKozak/AndroidAPS/issues/2868#issuecomment-679153252

You have been the only one who is mentioning this so far (after the release of 2.7) so maybe I was wrong in thinking this behaviour would get quite some attention. Also see: https://github.com/MilosKozak/AndroidAPS/issues/2924

maheini35, 2020-10-12:
> 
> 
> This is not a bug. You can read about it here:
> [MilosKozak/AndroidAPS#2868 (comment)](https://github.com/MilosKozak/AndroidAPS/issues/2868#issuecomment-678638468)
> [MilosKozak/AndroidAPS#2868 (comment)](https://github.com/MilosKozak/AndroidAPS/issues/2868#issuecomment-679153252)
> 
> You have been the only one who is mentioning this so far (after the release of 2.7) so maybe I was wrong in thinking this behaviour would get quite some attention. Also see: [MilosKozak/AndroidAPS#2924](https://github.com/MilosKozak/AndroidAPS/issues/2924)

Thanks mate. Will read this and change my bug afterwards

wootmasterslick, 2020-10-14:
Would be nice if carb and insulin absorption calculation continued irrespective of BG values coming in. Now we often have that during a sensor swap the carbs are displayed as n/a because no BG values come in.

MilosKozak, 2020-10-15:
> Would be nice if carb and insulin absorption calculation continued irrespective of BG values coming in. Now we often have that during a sensor swap the carbs are displayed as n/a because no BG values come in.

hmmm it's calculated from BG values

doraberta, 2020-10-21:
I'm sorry to have opened the post under the old repo (see the link from ArthurusDent) , so my last answer here again: 

I read the threat about the defective Libre sensor.
This is about a torn out sensor, which then sends values that are too low over a long period of time. For this particular situation, that's fine.

The message that I opened is about values that are too high and will not be corrected. Here, in my opinion, a distinction can be made, since sensors that are torn out never send values that are too high.

From my point of view, the algorithm should be changed / corrected so
a) that with a flatline of low values the existing 45 min algorithm takes effect.
b) With high values (= in my opinion all> 80 BG), the algorithm must not work. = standard work by AAPS

ToreBj, 2020-10-21:
Adding my comment from the old repo here:

Is there a link to this Libre event?

I question the general safety precaution introduced for all CGMs caused by this (one?) event.
One should not compare the Libre with the Dexcom. A dexcom G5/G6 in native mode has a lot of diagnostics enabled and would capture most failure scenarios. I know from the time I used G5 and could easily see that the native mode would make the sensor fail, but turning off the native mode you could continue reading "raw" values.
The same goes for Libre where you are actually collecting raw values and therefore bypassing any built-in diagnostics.

For Libre reference see: https://www.facebook.com/groups/LoopedUK/permalink/367511433884633

"After the swim, Spike read super high at 22 mmol and LOOP enacted a high temp basal. The Libre reader itself said ‘too cold to read’, and is still not reading 2 hours later. (I asssume the sensor is dead, still reading sensor error) However Spike was still getting readings at 22mmol and transmitting to LOOP to act on it."

I trust my Dexcom G6 and would like to lower my BG in high BG situations. I've seen this precaution acting personally many times and propose to disable this for all "safe" CGMs (as with SMB, i.e. enabling SMB only for "safe" CGMs).

MilosKozak, 2020-10-25:
changed for Dexcom native app

closed 2020-10-25 by MilosKozak

ArthurusDent, 2020-10-25:
Is it possible to also change this for xDrip+ with Dexcom?

dv1, 2020-12-03:
> Is it possible to also change this for xDrip+ with Dexcom?

I'd like to know that as well. If xDrip is used with native mode, it getting the same BG data the official app is receiving, and the transmitter will intervene if the values seem bogus.