nightscout/AndroidAPS #36: Feature request: Do not hide insulin button

cascer1, 2020-10-14:
when the app cannot connect to my pump (or is still initalizing) the insulin button is hidden.

Sometimes I manually inject insulin which I wish to enter using the "record only" feature. However, this is impossible when the insulin button is hidden.

Instead, would it be possible to disable submitting the insulin form when attempting to bolus when there is no connection?

MilosKozak, 2020-10-15:
adding exceptions means more possible bugs. i don't think this is usefull for most poeple

closed 2020-10-15 by MilosKozak

StephenBrown2, 2020-10-15:
This would be useful for me as well. The current workaround of waiting till the pump connects, then using the insulin button and altering the time to record an entry does work, though I have to note the time and remember to enter it later.

cascer1, 2020-10-16:
> adding exceptions means more possible bugs.

By this logic we should never introduce any features to any code because they might introduce bugs. That's simply the result of writing software.

Couldn't you at least listen to what other people think before dictating that a feature is not worth implementing? @MilosKozak