nightscout/AndroidAPS #44: Merge Buttons Insulin + Bolus (wish / suggestion)

doraberta, 2020-10-21:
(I hope I'm right here with my wish) 
I think it would be good if the two buttons "Insulin" and the additional "Bolus" could be combined in one. There are not many differences, only "Eating Soon-TT", "carbohydrates" and the "insulin amounts to be pre-assigned".
Background:
I miss the insulin defaults in the bolus button and the carbohydrates in the insulin button.

Rgds
Dirk

MilosKozak, 2020-10-25:
what is additional Bolus ?

doraberta, 2020-10-26:
Hi Milos,
sorry for the bad explanation. I mean these both buttons:
[image]
[image]

MilosKozak, 2020-10-26:
integrating everything into one button would lead to dialog with too many option. and this is something I don't want for most used dialogs.
combined button was requested by people who are not using calculator to be able enter both at once

so I don't plan to change it

closed 2020-10-26 by MilosKozak