nightscout/AndroidAPS #46: AAPS not sending basal to xDrip

JimZam, 2020-10-22:
Good evening. Since I updated to AAPS version 2.7 xdrip doesn't receive basals. I have Local Braodcast enabled in AAPS and Allow treatments enabled in xDrip, and it receives carbs and bolus but not basals. Any help is welcome.
Thank you in advance and regards

Edit: xDrip is receiving cabs and bolus from Nightscout, not AAPS. If I disable data (wifi and 4G) no bolus or carbs is received even with NSClient local broadcasting. I don't know what I'm doing wrong...

Amidamaru, 2020-10-22:
I have the same issue.
in advance: sorry, for hijacking, but it seems and general issue.
xDrip is not getting any data/treatment whatelse from AAPS v2.7 -> the only thing which is working is to push calibration from AAPS to xDrip.
in AAPS I've tried setting up NS without an valid URL ( since I am not using NS and I won't in near future hehe), because only then the extended setting is listed: "Local Broadcast". But it seems this local broadcast does not really broadcast anything by itself to xDrip/local.
Imho it would be better to have this option "local broadcast" separated from NS plugin. Any information on that, why local broadcast is within the NS plugin?

Tornado-Tim, 2020-10-26:
Enable Xdrip Statusline plugin from the configBuilder

ArthurusDent, 2020-10-26:
@Tornado-Tim Ok, that's at least something. The status line does show the current basal rate, even when I do not have internet access. That means AndroidAPS sends the data successfully and xdrip receives it. But on my phone xdrip still does not draw the basal rate in blue at the top of the screen. "Show basal TBR" is enabled. 

It's a weird problem and I remember that several people have that problem. Looking at xdrip's issue tracker there does not seem to exist an issue for this problem, though. I'll try to create an issue in the next few days and post a link to it here.

JimZam, 2020-10-27:
I did it there at the same time as here
https://github.com/NightscoutFoundation/xDrip/issues/1481

teleriddler, 2020-11-01:
Also having the exact same issue (and have had it since Dec of last year).  I mostly live in AAPS so I monitor my basal from there and Nightscout successfully plots my basal as well.  I have tried all recommendations from the Gitter room to no avail.  This is the first I have heard of trying to turn on the xDrip+ statusline in the config builder.  I have done this, restarted both apps and still not basal plots.  Curious to see the outcome of this issue.

My setup:

Samsung Galaxy S7 Edge - Android 8.1
AAPS 2.7 (AAPS-Omni branch)
xDrip+ nightly build from 01.10.2020
Omnipod Eros
Libre 2 (with patched app)

andyrozman, 2020-12-14:
This worked in very old AAPS (1.5.x I think), but I haven't seen any of my basal data in Xdrip+ in last 3 years. But I can't say exactly when it stopped working.

JimZam, 2020-12-14:
It's strange, it works since a couple of weeks ago, in my old phone and in my new one (installed everything from scratch three days ago)

AdrianLxM, 2020-12-15:
AAPS never sent basals to xDrip.
xDrip was parsing those from the watch data. That was about to fail.

We could send the data in any form needed that it doesn't need to be parsed.

ArthurusDent, 2020-12-21:
I would like to remind everybody that this issue is about at least two different problems/bugs.

The first one is the one mentioned in the headline: xdrip doesn't receive basals.

The second one is that xdrip doesn't receive bolus and carb entries once there is no connection to the internet. This has only been mentioned in the edited version of the first post.

It looks like the first problem could be worked on now.

Concerning the second problem, 4RK4N on the xdrip gitter channel might have found the problem: to him it looks like nsclient stops broadcasting locally once there is no internet connection available. This is of course not how this should work. 

Hopefully this will help one of the devs to fix this problem.

PedanticAvenger, 2020-12-22:
I particularly noticed this issue when I re-installed and tried to cleanup information flows when I was setting up tidepool.  My current setup involved NOT having xDrip talking to nightscout at all and when I broke that connection I lost all display of treatments in xDrip.  I'm willing to entertain the possibility that I've misconfigured something but I've iterated a LOT on this. Otherwise everything else works fine, xDrip sends BG to AAPS fine and uploads to tidepool fine, AAPS sends everything to nightscout fine too.
Basal display used to work for me in xDip but that was back in OpenAPS days, from other discussions I've had on the xDrip side I am not confident it's worked directly with AAPS but only as a result of other plugins that have massaged the data to make it appear to xDrip like something it previously expected.

closed 2021-01-17 by AdrianLxM

DiabetesPapa, 2022-02-28:
Basals are shown in xdrip+ when AAPS configuration - general - xdrip-status (watch) is active. Just found it after hours of searching. I hope the part about configuration, ... is correct, I use german language and translated freely :)
Might be old but still happens.

LuisPuertollano, 2022-02-28:
a screeshot maybe?