nightscout/AndroidAPS #47: App crashes when importing preferences

bartsopers, 2020-10-22:
The app crashes when importing preferences. I reverted some recent commits and found that this crash was introduced in commit 955844f3191d6c3aae5a3dcb91b17b929e66f479.

**Version**
2.7.0-dev (commit 492bb349ddb57d935118c7750674fade594ee347)

**Reproduction steps**
1. Go to Maintenance
2. Click Import Settings
3. App should now crash

**Stack trace**
```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: info.nightscout.androidaps, PID: 9854
    java.lang.IllegalStateException: LifecycleOwner info.nightscout.androidaps.activities.SingleFragmentActivity@dcb16f9 is attempting to register while current state is RESUMED. LifecycleOwners must call register before they are STARTED.
        at androidx.activity.result.ActivityResultRegistry.register(ActivityResultRegistry.java:116)
        at androidx.activity.ComponentActivity.registerForActivityResult(ComponentActivity.java:653)
        at androidx.activity.ComponentActivity.registerForActivityResult(ComponentActivity.java:662)
        at info.nightscout.androidaps.plugins.general.maintenance.ImportExportPrefs.importSharedPreferences(ImportExportPrefs.kt:264)
        at info.nightscout.androidaps.plugins.general.maintenance.ImportExportPrefs.importSharedPreferences(ImportExportPrefs.kt:259)
        at info.nightscout.androidaps.plugins.general.maintenance.MaintenanceFragment$onViewCreated$5.onClick(MaintenanceFragment.kt:55)
        at android.view.View.performClick(View.java:7339)
        at android.widget.TextView.performClick(TextView.java:14275)
        at android.view.View.performClickInternal(View.java:7305)
        at android.view.View.access$3200(View.java:846)
        at android.view.View$PerformClick.run(View.java:27787)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7078)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
```

AdrianLxM, 2020-10-23:
This actually is a bug that goes quite deep. It derives from an Android Jetpack bug that allowed this situation to happen before - but was fixed in recent versions. (https://issuetracker.google.com/issues/165435866)

From the documentation:
https://developer.android.com/training/basics/intents/result#register
> ...the callback must be **unconditionally** registered every time your activity is created, even if the logic of launching the other activity only happens based on user input or other business logic.

We don't meet the **unconditionally** criteria. We only register when needed.

Therefore in `ImportExportPrefs.kt` we would need to split this:
```
    fun importSharedPreferences(activity: FragmentActivity) {
        val callForPrefFile = activity.registerForActivityResult(PrefsFileContract()) {
            it?.let {
                importSharedPreferences(activity, it)
            }
        }

        try {
            callForPrefFile.launch(null)
        } catch (e: IllegalArgumentException) {
            // this exception happens on some early implementations of ActivityResult contracts
            // when registered and called for the second time
            ToastUtils.errorToast(activity, resourceHelper.gs(R.string.goto_main_try_again))
            log.error(TAG, "Internal android framework exception", e)
        }
    }
```

The registering part would have to be done in `onCreate()` or better in initialization of the activity.

@MilosKozak @TebbeUbben: can the maintenance fragment only be in the main-activity? If not, we might have to check if we can register for the fragment and not the activity or we could come into the situation to double-register :/

MilosKozak, 2020-10-24:
if this is easiest solution we can force always visible
@AdrianLxM

MilosKozak, 2020-10-26:
fixed by https://github.com/nightscout/AndroidAPS/commit/13a55eeb42a5b8f0914cc21f9292adba8518ef27

closed 2020-10-26 by MilosKozak