nightscout/AndroidAPS #50: [AAPS 2.7.0] -> temp target finish: target stuck on selected value with green background

mtothearkus, 2020-10-24:
I had already several times the following issue:

For doing sports I change profile to 30-40 % for example 3 hours (2 hours before sports, 1 during activity). I also put an temp target at 160 for not getting SMB and having a higher level during acitvity.

After sports, when I got home, i noticed that the profile changed successfully back and the temp target was also finished. But instead of going back to profile target (in my case 100) AAPS but remains on the target of the temp target before. Also it is green and not yellow anymore. 

I have the the new options for raising/lowering target regarding AS active, but 160 at AS 100 is way to high so I assume a bug.
When I set a TT and cancel afterwards, the correct 100 (without background color) is reset.

I use AAPS 2.7.0
Sony XZ2 Compact 
Android 9
Accu Check Combo
Dexcom G6 (patched app)

[image]

snip from log:
18:33:40.504 [main] D/CORE: [TempTargetDialog$submit$$inlined$let$lambda$1.run():135]: USER ENTRY: TEMP TARGET 160.0 duration: 60
19:33:31.923 [main] D/CORE: [OverviewFragment.updateGUI():661]: Adjusted target. Profile: 100.0 APS: 160.0

I also attached the log for the relevant time

[AndroidAPS._2020-10-20_00-00-13_.7.txt](https://github.com/nightscout/AndroidAPS/files/5432823/AndroidAPS._2020-10-20_00-00-13_.7.txt)

MilosKozak, 2020-10-25:
did you read this? 
https://androidaps.readthedocs.io/en/latest/EN/Getting-Started/Screenshots.html#visualization-of-dynamic-target-adjustment

ArthurusDent, 2020-10-25:
I've seen this, too, but the high temp target value on green ground is only shown for a few minutes at the end of the temp target duration. After that it switches back to the correct set target in the profile without the green background. I've only looked at this as a cosmetic bug.

NickSills1, 2020-10-25:
@mtothearkus I've seen similar, but for me it is caused by communication lost with the DanaRS pump at the end of the temp target time. When this happens,  the temp target remains 'stuck', as in your screen shot, until communication with the pump is successful. 

I had to restart the phone last time this happened. 

[image]

mtothearkus, 2020-10-26:
> did you read this?
> https://androidaps.readthedocs.io/en/latest/EN/Getting-Started/Screenshots.html#visualization-of-dynamic-target-adjustment

Yes and this works good - but as you can see in the screenshot AS is at 100 % (because of profile change before) and an adjustment of 160 for AS is way to high. 
e.g. with AS 90 % the target goes to 116 in my case. 

This might be just a cosmetic (don’t wanna test it), but is different to 2.6.1

MilosKozak, 2020-10-26:
this is 2.7 feature

MilosKozak, 2020-10-26:
it stay green until new OAPS run (without TT adjustment)

closed 2020-10-26 by MilosKozak

NickSills1, 2020-10-26:
> it stay green until new OAPS run (without TT adjustment)

So, in the absence of another TT adjustment, the current TT finish time rounds up to the next OAPS 5 minutes run. As long as it's understood and doesn't cause any problems, then there's no change required.  I guess that changing the target midway through the OAPS cycle has no effect, but It's a stretch to call this a 'feature'.