nightscout/AndroidAPS #51: Sony Smartwatch INSTALL_FAILED_OLDER_SDK

JimZam, 2020-10-24:
Good evening. When trying to install the app in may Sony Smartwatch 3 I get this error: "INSTALL_FAILED_OLDER_SDK" probably due to the watch not using Wear OS 2, I'm not sure. I can use the app from AAPS 2.6.4 but I've read in AndroidAPS documentation that "When using wear version of AAPS, always update it together with phone version of app - keep their versions in sync."

https://androidaps.readthedocs.io/en/latest/EN/Configuration/Watchfaces.html

What problems could I have if the version is not the same? Any way to compile the app so it works with my watch?
Thank you so much
Regards

MilosKozak, 2020-10-26:
watch has too old android version

closed 2020-10-26 by MilosKozak

JimZam, 2020-10-26:
Sorry to insist. Will my wear app stop working with the phone app?
Thank you again

an-di-looper, 2020-11-09:
I use the AAPS wear app v2.6 on Sony Smartwatch 3 and AAPS smartphone app v2.7 and have not noticed any issues. The only problem I see is that many many loopers are still using a SWR50 and since v2.7 they have to install old wear app via adb