nightscout/AndroidAPS #58: Import Settings: Android system notification to allow file access not visible when requested on import, need to hit back to see it on previous screen

teleriddler, 2020-10-30:
**Version:** 2.7.0
**Android Version:** 8.1
**Phone:** Samsung Galaxy S7 Edge

**NOTE:** I reported this issue in the AAPS-Omni branch, however Bart Sopers closed the issue as he said it is a core AAPS issue and not the AAPS-Omni branch specific, so I am reporting it here.

**Time:**  Issue is not dependent on time and an error does not appear in the logs.  UX issue.

**ISSUE:**  Upon a new installation of AndroidAPS, when importing backed up settings the first time an Android system notification is needed to allow access to the file system to retrieve a scan of any saved preferences files on the system.   The problem is that upon clicking "Import settings" under the maintenance tab you are taken to a new screen titled "Select file to import".  However there are not any buttons or items to execute on this screen.  Only upon using the system "back" button on the phone does AndroidAPS take you to the previous screen where the Android system notification for file access finally appears.  Clicking "allow" and then "Import Settings" again allows AndroidAPS to successfully scan the filesystem for preferences files and the user can choose one to continue with the import procedure.

**Steps to Reproduce:**

1. Backup existing AAPS preferences using 2.7.0 version of AAPS using a master password.
2. Uninstall existing version of AAPS
3. Install new version of AAPS
4. After installation, AAPS will immediately open.  Exit settings Wizard and force close the app.
5. Copy engineering_mode file to AAPS installation directory
6. Manually open AAPS
7. Go to 3 button menu -> Preferences -> General and set a Master Password.
8. Go to Hamburger menu -> Maintenance
9. Click Import Settings
10. You are taken to the Select file to import screen, however no options are available
[image]

11. Click the back button on your phone or the back arrow in the upper left corner.
12. On the previous Maintenance screen you will see the Android system notification asking for system file access.
[image]

13. Click allow
14. Click Import Settings again
15. You should now see all preferences files on your system, click one
[image]

16. On the Import Settings dialog ensure the filename you clicked matches what is shown in the dialog, if so, click OK.
[image]

17. Enter your master password and click OK.
[image]

18. If your password is correct the Import settings dialog will show a summary of your preferences import.  Click Import to complete the process.
[image]

19.  Click OK on the successful settings imported dialog to restart AAPS.
[image]

MilosKozak, 2020-11-07:
does the problem appear when go through wizard?

MilosKozak, 2020-11-12:
fixed by https://github.com/nightscout/AndroidAPS/commit/d1009e53218e90aac17639ff876b13467057ef4a

closed 2020-11-12 by MilosKozak

teleriddler, 2020-11-12:
Yes, both through the wizard and without the wizard.

I just tried with the latest commit and the problem has been resolved.  Thank you.