nightscout/AndroidAPS #63: Accu-chek Insight pump error M23 automatically stopped

PimSPIE, 2020-11-10:
**Version:** 2.7.0
**Android Version:** 9.0
**Phone:** OnePlus 6
**Pump:** Accu-check Insight firmware version 2.0.1
**FGM:** Freestyle libre 1 with MM2, using xDrip+

**Time:** the error occurs on average every 90 minutes

**Background:** 
First time user of AndroidAPS and looping in general since last sunday, November 8th. Followed the documentation for installation and all settings. Using a local profile, not set up a nightscout site yet. Using open loop for now. 

**Issue:**
At random times the pump starts to make noise and it gives the error "maintenance error M23 automatically pump stopped". At that point the pump is stopped and needs to be started again. The AndroidAPS app does not give an alert or anything. When manually refreshing the pumps connection it does show the information in graph and careportal. It shows the "automatically stopped" and then "pump started".
I'm not sure if this only happens when the open loop is changing the basal. 
Bolus is working without problems. Changing the basal manually works without problems.
Issue also occurs when looping is turned off!
Whats also a bit weird is that we changed sound/volume settings in the pump(as recommended in docs) but the pump still makes sound.

**What did I try:**
- New battery and new battery cap
- Repairing the pump with phone (using the AndroidAPS wizard), twice already. So removed the bluetooth connection in phone, in app and in pump. Restarted phone and pump, then repaired in app. Note: the pump does show up in bluetooth settings of the phone after this. 
- Disabled the firmware 3.x specific settings
- Searched for information in the documentation, in the devices list and on facebook groups. 

I hope this problem can be solved, because currently its almost unusable. 
If more information is needed please let me know.

TebbeUbben, 2020-11-11:
It seems like you have enabled the pump's "Automatic Off" function. Please refer to section 8.4.1 of the manual to find out how to deactivate it: https://www.accu-chek.co.uk/download/file/fid/28316

closed 2020-11-11 by TebbeUbben

PimSPIE, 2020-11-11:
@TebbeUbben  yesterday I found out I was indeed searching on the wrong places. Found that manual yesterday and the function was indeed the cause of the problem. So its nice that its all solved now !
Question: can the AndroidAPS app change the setting of that function ? 
Why am I asking this: I never knew about this function before, never used it and never changed its setting. It started doing this after connecting with AndroidAPS.

TebbeUbben, 2020-11-11:
The corresponding packet structure has not been implemented, so it is very unlikely - if not impossible - that AndroidAPS has changed this setting.