nightscout/AndroidAPS #71: AAPS Random restart since upgrade to V2.7

bicomm, 2020-11-14:
hi,
Since upgrade to AAPS 2.7, the app restarts on its own a few times a day ramdomly.  
It happens whether i'm using an other app on phone Or NOT
In same environement, i had no issues with 2.6.1 AAPS version. 

System environment:
phone: Xiaomi MI9T
Android version: 10 / MIUI 12.0.2
Pump: Dana RS (fimrware V1) 
CGM: FSL+MIAO1
[AndroidAPS._2020-11-06_18-30-37_.log](https://github.com/nightscout/AndroidAPS/files/5541444/AndroidAPS._2020-11-06_18-30-37_.log)

in log file, first iteration happens on line 8408. 

--------------

MilosKozak, 2020-11-22:
sorry the reason is not visible in log files.
Lot of small fixes are included in dev. You may test it

closed 2020-11-22 by MilosKozak

nickxbs, 2021-01-03:
hi,
same problem for me that still persists 2.8
Nothing in the aaps logs, but in android's logcat there is an AutoPowerKill that kills the app. 

[logcat.log](https://github.com/nightscout/AndroidAPS/files/5762026/logcat.log)

(All power optimization and notification settings are turned of for AAPS)

Xiaomi readmi note 7
Android 10
AAps 2.8

MilosKozak, 2021-01-04:
it's phone specific. I cannot help.
Did you everything specific for your phone on pageaccessible from About dialog?

nickxbs, 2021-01-04:
yes i did! it seems to be impossible to manage. https://dontkillmyapp.com/xiaomi
...I will try to change the rom or come back miui11

Thank you @MilosKozak