nightscout/AndroidAPS #72: [2.7] connection to pump get lost

thomasmenges, 2020-11-14:
During the day and also night it could happen, after receiving a new BG value from rhe patched dexcom app that the pump connection get lost.

AAPS is stil waiting for a result from  the pump. New BG values are coming in but AAPS isn't activating a new connection to the pump and still waits for a not coming result of the pump.

Exit AAPS and starting AAPS again helps to solve the problem diretly.

Why can't AAPS recognize that the pump does'n't delivers results for SMB or temp basals after a defined time and restarts tAAPS ?

By the way the closed issue whichvdescribes my problem couldn't solve my problem.

I have already reinstalled AAPS for severeal times and all accu optimizations on my phone are switched off.

[image]

MilosKozak, 2020-11-16:
turn on pump debugging , and provide logs with time of occurence.  this not enough to be able to help you

thomasmenges, 2020-11-16:
Attached you can find the log.zip file.
Time is difficult to say because I recognize the problem only after some time when  temp basals or SMBs aren't done anymore.

Try to use 09.00  and some minutes before.

[AndroidAPS_LOG_1605308881804.log.zip](https://github.com/nightscout/AndroidAPS/files/5545208/AndroidAPS_LOG_1605308881804.log.zip)

MilosKozak, 2020-11-16:
logs doesn't conataing that time. do what said and provide logs on next occurence

thomasmenges, 2020-11-16:
Here is a new log file. Please use time 17:32

And 2 new screen shots

[AndroidAPS_LOG_1605544633665.log.zip](https://github.com/nightscout/AndroidAPS/files/5548400/AndroidAPS_LOG_1605544633665.log.zip)
[image]
[image]

MilosKozak, 2020-11-22:
Pump failed to set TBR and this state was not properly reported to loop tab
fixed by https://github.com/nightscout/AndroidAPS/commit/6f7cfdd06dba07907a8dd4781182d12e9d4424b5

closed 2020-11-22 by MilosKozak