nightscout/AndroidAPS #75: AAPS+Nightscout for selfhosting (SSL)

DOFfactory, 2020-11-20:
I am new to the whole Nightscout+AndroidAPS ecosystem. However, I would love to try it out properly. The thing is that I have a private server, into which all my devices all connected via VPN. 

On this server I would love to use Nightscout. I cloned it and I was successful in setting it up via docker. I am able to access the local and not publicly available website of Nightscout. I am really reluctant to make my Nightscout page publicly available, as I may willingly disclose my personal medical data (unwillingly I might violate also GDPR as a data-handler of my own private medical data), while also providing another vector of attack into my systems.

On the other end, AndroidAPS requires SSL connection to Nightscout, many options are simply not working without SSL. Currently AndroidAPS does not have the option to work neither with self-signed certificates, nor with disabling this requirement, as my VPN connection handles the security. I tried `engineering_mode` but I still get the error `NS URL not encrypted` (as discussed [here](https://gitter.im/MilosKozak/AndroidAPS?at=5e98840ed65bcf75b5c62918)). 

Would it be please possible to allow unencrypted connection from AndroidAPS to NS with all the options for those of us who are self-hosting? Unfortunately Nightscout does not provide full authentication to the main screen, which I am feeling uneasy about to share publicly. 

**TLDR**: Provide either full authentication to Nightscout or allow unencrypted connection from AndroidAPS to Nightscout without SSL for self-hosters.

Tornado-Tim, 2020-11-21:
Have you tried Lets encrypt. Open a basic webserver for auth then copy the key over to your private NS front end proxy or NS server directly.

DOFfactory, 2020-11-21:
Hi @Tornado-Tim, thanks for the follow-up. I am using NS from docker, and I allow access to my docker containers through Caddy with my domain. Caddy provides https by default. My issue is that even if I have https+domain+NS, whoever knows the address (whichever robot scans the web), the NS page is accessible to the world. 

Now, Caddy provides a basic auth solution: https://caddyserver.com/docs/caddyfile/directives/basicauth. Do you refer to this option? Do you think that the API secret can go through this authentication regardless?

Tornado-Tim, 2020-11-21:
In that case, you can use a certbot for wildcard cert and DNS auth which will allow you to get your certs for NS. Since its wildcard, your NS will not need to be exposed.

DOFfactory, 2020-11-21:
Yeah, this might work. Unfortunately I have a domain with GoDaddy, and the DNS auth module is not ready yet (no experience with the Go language, so I cannot write one myself). I was wondering, will AndroidAPS accept self-signed certificates? Or it must be from signed by entities like LE?

MilosKozak, 2020-11-21:
i'm not familiar with caddy but nginx has `allow` and `deny` directives for access filtering

rICTx-T1D, 2020-11-21:
When you like use a selfhosted webservice NS with private certificate, you must import the public key in android trust store. See android setting in security selection.

DOFfactory, 2020-11-21:
Yeah, these are all way too complex solutions compared to the single and simple one: there should be a login page for NS, after entering the credentials it should allow the user to the next page. Contacting the API with the API_SECRET is fine, I am not sure how it works in the background, and whether it should happen on another port. 

By this simple solution one can then expose the NS server with a reverse proxy solution - with https.

swissalpine, 2020-11-22:
I have a similar problem. AAPS can receive and send encrypted data to my selfhosted nightscout-docker, but AAPS cannot retain the connection. Instead of the "ping" I get an XHR poll error ... I have a Let's encrypt certificate for this subdomain and use a reverse proxy.

MilosKozak, 2020-11-22:
[image]
this is my nginx setting

DOFfactory, 2020-11-22:
OK, I made it working with the latest NS version (I used niepi/cgm-remote-monitor:14.0.7), there one needs to authenticate even for viewing the web frontend. With this I was able to open the site to the Internet.

FYI, I use Caddy for securing the web with https by default (I have a domain), if this helps, I am pasting the `Caddyfile` config, which is super simple:

```
ns.{$MY_DOMAIN} {
   reverse_proxy 192.168.0.40:1337
}
```

closed 2020-11-22 by DOFfactory

rICTx-T1D, 2020-11-22:
@swissalpine Maybe the problem can solved,  when you allow ping the public host ip adress from you gateway. Otherwise the backend is not communicating the correctly ip through to reverse proxy. On many configuration with apache tomcat and reverse proxy in front, the same issue is present. I don't know how nightscout in docker must be configure to inform NS that a reverse proxy is working in front.

swissalpine, 2020-11-22:
@MilosKozak Thanks. Your settings led me on the right path. I needed the following slightly modified header:
`proxy_set_header Connection $connection_upgrade`

wootmasterslick, 2022-07-22:
Honestly, I find it quite frustrating that it has to be so complicated to setup a local NS server to work with AAPS. Ideally there would be an option to push the data to a not SSL server on a closed off LAN and leave al the certificates out of it.

MilosKozak, 2022-07-22:
current libraries doesn't allow http only. that's it

wootmasterslick, 2022-07-22:
Its proving to be veryu hard to get up and running. I tried nightscout with https settings enabled and self signed certificates. get XHR poll error. Now have it setup nginx with same certificates, still the same problem. I imported the cert to the phone, still same problem. Tried disabling the https enforcement with engineering mode, didnt work. Tried to remove the code that checks for https, build failed.  I am getting lost here. Im in so deep now that is MUST get it to work or I'll go mad.

robertrub, 2022-09-07:
To add to this discussion. NS and Mongo db in a Docker container on my QNAP NAS. External 4443 routed to internal 1337. Letsencrypt certificate for the NAS. Reverse proxy from external htttps 4443 to internal http 1337
I can access locally in http via nasIP:1337 in xDrip and navigators. I can access from the Internet in https via mynasdns:4443 from xDrip and navigators. Only AAPS gives _XHR poll errors_ even though the NS terminal in the Docker container shows the connection to AAPS (no data is received by NS from AAPS).
Same setting in xDrip sends the data to NS correctly.

I wonder if it is the :port after the address that is causing the problem(wild guess, not tested as I can't direct 443 to 4443 on the router).

swissalpine, 2022-09-08:
No, I have nearly the same setup on my Synology NAS with the same behaviour: xDrip has no problems, AAPS throws errors.
With Milos hints (at the end of the thread) I was capable to fix this.

petervanrijt, 2022-09-14:
@wootmasterslick very interesing, have you got a pull request for this to review and possible implement in AAPS?

TiagoPRSilva, 2023-09-09:
> I managed to replace part of the nsclient in AAPS so as it accepts self signed certificates.

Hey there, can you please share what was the fix on aaps?

i am having a similar issue for quite some time and still didnt find a solution for this.

I have setup nightscout on an ubuntu home server.

My reverse-proxy is swag container from linuxserverio.

Everything works ok, only the nsclient on aaps cant communicate with nightscout.

thanks :)

MilosKozak, 2023-09-11:
like mentioned here. create selfsigned cert and add it to android

wootmasterslick, 2023-09-11:
> > I managed to replace part of the nsclient in AAPS so as it accepts self signed certificates.
> 
> 
> 
> Hey there, can you please share what was the fix on aaps?
> 
> 
> 
> i am having a similar issue for quite some time and still didnt find a solution for this.
> 
> 
> 
> I have setup nightscout on an ubuntu home server.
> 
> 
> 
> My reverse-proxy is swag container from linuxserverio.
> 
> 
> 
> Everything works ok, only the nsclient on aaps cant communicate with nightscout, returning the xhr poll error indefinitely.
> 
> 
> 
> thanks :)
> 
> 

Will have a look this week. I believe I changed some value in the AAPS code and that enabled it again to run using http. So no more certificates. I didn't get it to accept self signed certificates as those are checked by AAPS and found to be invalid.

setonez, 2023-11-11:
> like mentioned here. create selfsigned cert and add it to android

Since Android 7.0 user certificates are not trusted by default (https://developer.android.com/privacy-and-security/security-config#network-security-config). What I did is:

1. Install my self-singed certificate as a trusted user certificate in my phone
2. Configure AAPS to trust on user certificates as described in the previous link (add app/src/main/res/xml/network-security-config.xml file and modify AndroidManifest.xml with android:networkSecurityConfig="@xml/network_security_config")

Hope it helps.

old-square-eyes, 2024-01-03:
@setonez Any chance you can put in a bit more detail? I tried both adding my cert to the build per the linked instructions, and also allowing all user certs installed on Android per your suggestion (with "<certificates src="user"/>"). Neither worked.
[image]

petervanrijt, 2024-01-03:
@old-square-eyes I suggest to use a Let's Encrypt certificate (if that is not the case), works like a charm on our set-up.

wootmasterslick, 2024-01-03:
> @setonez Any chance you can put in a bit more detail? I tried both adding my cert directly to the build in `/res/raw` per the linked instructions, and also allowing all user certs installed on Android per your suggestion (with `<certificates src="user"/>`). Neither worked and had the same result in the screenshot below.
> 
> 
> 
> My cert is uploaded to Android as a .crt in pem format. When adding to the AAPS build I tried with and without file extension (only without linked properly). I tried with and without headers.
> 
> 
> 
> [image]
> 
> 
> 
> 

You can also modify the AAPS app code to accept a http server url in the nsclient fields (currently its locked to a https only url) then run unencrypted. Thats what I'm doing on my local network.

old-square-eyes, 2024-01-03:
Unfortunately I ha e CGNAT, so home server is not an option. Though,
current ISP is rubbish so might be time to find a better one.

old-square-eyes, 2024-01-04:
> I also tried getting selfsigned certs to work but see its still being validated against database even when it should be disabled via engineering_mode.

I don't have engineering mode enabled. Should I? I've just made the stated changes to network_security_config.xml

wootmasterslick, 2024-01-04:
> > I also tried getting selfsigned certs to work but see its still being validated against database even when it should be disabled via engineering_mode.
> 
> 
> 
> I don't have engineering mode enabled. Should I? I've just made the stated changes to network_security_config.xml

Give it a try

old-square-eyes, 2024-01-07:
> > > I also tried getting selfsigned certs to work but see its still being validated against database even when it should be disabled via engineering_mode.
> > 
> > 
> > I don't have engineering mode enabled. Should I? I've just made the stated changes to network_security_config.xml
> 
> Give it a try

Because it's going to work? I don't particularly want to have to enable engineering mode if I don't have to (safety and all that).

setonez, 2024-01-09:
> > @setonez Any chance you can put in a bit more detail? I tried both adding my cert directly to the build in `/res/raw` per the linked instructions, and also allowing all user certs installed on Android per your suggestion (with `<certificates src="user"/>`). Neither worked and had the same result in the screenshot below.
> > My cert is uploaded to Android as a .crt in pem format. When adding to the AAPS build I tried with and without file extension (only without linked properly). I tried with and without headers.
> > ![Screenshot_20240104_100036_AAPS](https://private-user-images.githubusercontent.com/25760491/294029647-f5881916-d6de-4bc4-8032-8a6c780c9239.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4MTU2MzYsIm5iZiI6MTcwNDgxNTMzNiwicGF0aCI6Ii8yNTc2MDQ5MS8yOTQwMjk2NDctZjU4ODE5MTYtZDZkZS00YmM0LTgwMzItOGE2Yzc4MGM5MjM5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA5VDE1NDg1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg3MTc2N2NhN2RhNWNjNDBmNzgzODVhNTIzNTkwM2ZiOWQ0ZjYyNTM5YTI4ODE1ZjI3ZTM0NGUzZThlMDQ3N2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BUdfMQqQomDHgDsx6--0O8qz7T7AQoVgPLzbRjGRfQg)
> 
> I also tried getting selfsigned certs to work but see its still being validated against database even when it should be disabled via engineering_mode. You can also modify the AAPS app code to accept a http server url in the nsclient fields (currently its locked to a https only url) then run unencrypted. Thats what I'm doing on my local network.

Hi @old-square-eyes, sorry for late reply.

These are all the changes I've made in AndroidAPS:
1. Modify **AndroidManifest.xml**
   ```
    diff --git a/app/src/main/AndroidManifest.xml b/app/src/main/AndroidManifest.xml
    index 6baa3a5a34..a898e90f3f 100644
    --- a/app/src/main/AndroidManifest.xml
    +++ b/app/src/main/AndroidManifest.xml
    @@ -43,6 +43,7 @@
             android:restoreAnyVersion="true"
             android:roundIcon="${appIconRound}"
             android:supportsRtl="true"
    +        android:networkSecurityConfig="@xml/network_security_config"
             android:theme="@style/AppTheme.Launcher">
    
             <meta-data
   ```
1. Add a new file **app/src/main/res/xml/network_security_config.xml** 
   ```
   <?xml version="1.0" encoding="utf-8"?>
   <network-security-config>
       <base-config>
           <trust-anchors>
               <certificates src="system"/>
               <certificates src="user" />
           </trust-anchors>
       </base-config>
   </network-security-config>
   ```
1. Build the apk and install in my phone
2. Install the root certificate in my phone under  `Security Settings --> More Security Settings --> Install certificates from storage`

If you problem persists I guess it might be related with your certificates. There are many guides describing how to do it, adding here the one I use, maybe it helps you: https://www.digitalocean.com/community/tutorials/how-to-set-up-and-configure-a-certificate-authority-ca-on-ubuntu-20-04

old-square-eyes, 2024-01-10:
> ere the one I use, maybe it helps you:

Thanks for the reply. I have done everything you have. I used a different tutorial to create the cert. The cert itself works fine when browsing (after allowing the exception). I'm stumped. Do you have engineering mode enabled?

setonez, 2024-01-10:
> eering mode enabled?

No, I don't.
Quick google search points to configure Subject Alternative Name (SAN) in the certificate... I configured it for my local IP...

old-square-eyes, 2024-01-10:
> Subject Alternative Name

Finally got it to work. Thanks for the pointer. Seems SAN _is_ required.

old-square-eyes, 2024-01-10:
`#!/bin/bash

# This bash script generates, places and tests self signed certs for self hosted sites.
# Be sure you are aware of potential risks involved with allowing self signed certs, and customising AAPS. No warranty provided.

# Configurable Parameters
local_domain="nightscout.home"
cert_dir="/etc/ssl/certs"
key_dir="/etc/ssl/private"
home_dir="$HOME"
country="NZ"   # Update as needed
state="Auckland"   # Update as needed
locality="Auckland"   # Update as needed

# Check if directories exist
[ -d "${cert_dir}" ] || { echo "Certificate directory not found: ${cert_dir}"; exit 1; }
[ -d "${key_dir}" ] || { echo "Key directory not found: ${key_dir}"; exit 1; }

# Generate a new private key and save it to the NGINX dir
if ! openssl genrsa -out "${key_dir}/${local_domain}.key" 2048; then
    echo "Failed to generate private key."
    exit 1
fi

# Generate a new certificate and save it to the NGINX dir
if ! openssl req -new -x509 -key "${key_dir}/${local_domain}.key" \
  -out "${cert_dir}/${local_domain}.crt" -days 365 \
  -subj "/C=${country}/ST=${state}/L=${locality}/O=Nightscout/CN=${local_domain}" \
  -reqexts v3_req -reqexts SAN -extensions SAN \
  -config \
  <(echo -e "
    [req]
    distinguished_name=req_distinguished_name
    [req_distinguished_name]
    [SAN]
    subjectKeyIdentifier=hash
    authorityKeyIdentifier=keyid:always,issuer:always
    basicConstraints=CA:TRUE
    subjectAltName=DNS:${local_domain}
  "); then
    echo "Failed to generate certificate."
    exit 1
fi

# Convert the certificate to DER format and save it to the home directory for use in your device
if ! openssl x509 -in "${cert_dir}/${local_domain}.crt" -outform DER -out "${home_dir}/${local_domain}.der.crt"; then
    echo "Failed to convert certificate to DER format."
    exit 1
fi

# Output and Instructions
echo "Certificates and key generated:"
echo "  PEM Certificate: ${cert_dir}/${local_domain}.crt"
echo "  Private Key: ${key_dir}/${local_domain}.key"
echo "  DER Certificate: ${home_dir}/${local_domain}.der.crt"
echo "Optional:"
echo "  Inspect and validate your cert with: openssl x509 -in ${cert_dir}/${local_domain}.crt -text -noout" 
echo "  Review/edit NGINX config with:  nano /etc/nginx/sites-available/nightscout (where \"nightscout\" is your site)"
echo "  Test the NGINX configuration for syntax errors: nginx -t"
echo "  Copy the .der cert to your local machine for installing on your phone e.g. scp nightscout:~/${local_domain}.der.crt ~/"
echo "  Original ssl tutorial: https://medium.com/@noumaan/ssl-app-dev-a2923d5113c6"
echo "  Original NS self host tutorial: https://gist.github.com/johnmales/1b3c927f2a56aae640b4b2cd0298b1e7"

# Restarting NGINX
echo "Restarting NGINX"
if service nginx restart; then
    echo "NGINX restarted successfully."
else
    echo "Failed to restart NGINX. Run optional checks above."
    exit 1  # Exit the script with an error status
fi
`

sircsaba, 2024-05-17:
wootmasterslick, Can you please tell me exactly where I can modify the AAPS code to accept HTTP for my self hosted NS ? (Which file and where and what etc ...) NS is installed on a RPI4 with docker. NS Version is 14.2.6