nightscout/AndroidAPS #80: Insight Pump data not shown on Nightscout

DOFfactory, 2020-11-23:
I am not sure whether this is an AAPS or NS problem, so I start with AAPS. It makes the connection to the Accu-Chek Insight pump, collects data, the NS Client is working and uploading data. Except the pump-related data. Tried to check the logs, but could not find anything meaningful (ie. error) there. 

I am using `niepi/cgm-remote-monitor:14.0.7` in docker container, with the following pump-relevant ENV settings:

```
ENABLE=careportal boluscalc basal food rawbg iob maker bridge bwp bage cage iage sage pushover treatmentnotify loop pump profile openaps override dbsize
DEVICESTATUS_ADVANCED=true
PUMP_FIELDS=battery reservoir clock status device
PUMP_RETRO_FIELDS=battery reservoir clock status device
SHOW_FORECAST=openaps
SHOW_PLUGINS=openaps pump iob sage cage careportal
PUMP_ENABLE_ALERTS=true
PUMP_URGENT_BATT_V=1.3
PUMP_URGENT_RES=10
PUMP_URGENT_CLOCK=30
```

All the other data are being properly uploaded, except the pump-related ones:
[image]

Do any of you have any similar experience? Is the link between Insight <-> NS broken?

MilosKozak, 2020-11-24:
check records in mongo database
https://yourns/api/v1/devicestatus.json

DOFfactory, 2020-11-26:
I am getting this: 
```
{"status":401,"message":"Unauthorized","description":"Invalid/Missing"}
```

MilosKozak, 2020-11-26:
hmmm enable default read access or look directly into mongo database

DOFfactory, 2020-11-26:
I am already admin, using the standard API_SECRET in the browser.... did you have in mind something else? My AndroidAPS is logged in as admin, so should have all the access too

DOFfactory, 2020-11-26:
OK, I think I figured it out. So I set up my AndroidAPS' NS Client properly, the connection was working (please note I have also a VPN enabled). All the data have been uploaded, except the pump data. 

Now, things changed, when I rebooted the phone, the data from the pump appeared in NS. Hope this info is helpful. It would be nice if AndroidAPS could survive a VPN change, and the connection on the NS Client needs to be investigated thoroughly so no reboot is needed next time.

DOFfactory, 2020-11-26:
Just a follow-up, I have an exception for Android APS to avoid the VPN connection and route outside of it, however, AndroidAPS somehow does not respect this setup. Using this app for VPN: https://play.google.com/store/apps/details?id=de.blinkt.openvpn&hl=en

MilosKozak, 2020-11-27:
what's the problem?

DOFfactory, 2020-11-27:
The problem is the following: I am using NS+mongo on my self-hosted rig, instead of Heroku+github and the other solutions. I have an OpenVPN access to my server, and some services I connect to through the Internet with my own domain name (example.com).

I initially thought that I can set up NS+AAPS+CGM (Bubble's [DiaBox](https://www.bubblan.org/)) where all the data will be loaded into NS with the OpenVPN connection from my phone (not using xDrip+, 2 apps are enough for my battery life). The problem is that NS requires a valid SSL certificate to the server, which I am not able to obtain in my LAN (DNS challenge [does not work yet with GoDaddy and Caddy v2](https://github.com/caddy-dns)). 

This forced me to open NS to the Internet. However, for some reason, DiaBox does not upload data to NS through `https`, so I have to VPN back home and use a LAN address to do it successfully (emailed the developers, waiting for their reply). NS cannot do this (see above), so I have to route it through the Internet in order to do this through `https`.  [Blinkt's OpenVPN app](https://play.google.com/store/apps/details?id=de.blinkt.openvpn&hl=en) can do the separation, so AndroidAPS connects through normal Wifi, all other connections go through OpenVPN. The connections are successful (I see all the pings in the logs), all the data are uploaded to NS, except one, and that is the pump status. **AndroidAPS uploads all the profiles, treatments, etc., to NS except the pump data. I guess this is some bug.** As soon as I disconnect from VPN and restart the phone, the pump data is uploaded correctly, but in this case Diabox cannot upload the BG results to NS. 

The best option would be to allow for cases of selfhosting, i.e. for those who are non-stop connected to their self-hosted devices through VPN, to disable the requirement for `https` in AndroidAPS, as the connection **_is_ secured**, just not how it expects it.

MilosKozak, 2020-11-27:
i believe i cannot help. latest library used in AAPS requires https.

closed 2020-11-27 by MilosKozak

DOFfactory, 2020-11-27:
So you do not find it weird that all data is uploaded except pump data? :D

MilosKozak, 2020-11-27:
how do think an app can affect internal android routing?

DOFfactory, 2020-11-28:
OK, I just want to help you finding the issue. Could you please check again the initial screenshot I posted and explain me why all the other data is loaded from AAPS except the pump data? This data is available in AAPS, I have no explanation why it is not being uploaded to NS, while the connection was established "2 minutes ago" ... Again, all the basals, boluses, etc. is updated correctly from AAPS -> NS.

MilosKozak, 2020-11-28:
pump data is provided by pump driver. i can imagine situations like pump not initialized where pump data is not provided by driver

DOFfactory, 2020-11-28:
How can I check this? The bluetooth connection is made to Insight fw v2.0. The phone receives data from the pump as far as I can see.

MilosKozak, 2020-11-28:
whats in devicestatus jsons?

DOFfactory, 2020-11-28:
> I am getting this:
> 
> ```
> {"status":401,"message":"Unauthorized","description":"Invalid/Missing"}
> ```
> 
> Please note that I am logged in in the tab on the basic page https://myns.com, however, when I add the part `/api/v1/devicestatus.json` it does not work.