nightscout/AndroidAPS #93: Wish for Automation

doraberta, 2020-12-03:
Hello,
I love the automation in AAPS on the other hand I consider it very dangerous. It is recommended to create an abort for every automation, but it is not mandatory.
Background: 
I am very sensitive to insulin, but have various fluctuations in my BR, which I can only compensate by changing BR. The fluctuations are only 2-10%, but Autosense cannot compensate for these few percent. (A 2% difference means a value of 60 or 200 without AAPS in the morning). Therefore, I either work with a change in the BR, whereby IC + ISF remain constant, or only with an increase in ISF (BR and IC remain constant)
In order to only compensate for slight increases in the short term, I increase the BR for a short time by 20% and then reset it to the original BR with another cancellation rule. And here is part of my problem because you always have to select a defined BR for the cancellation rule. (Conversely, if I choose a different BR, I also have to adjust the automation at the same time)

I would like the following:
a) Selection of the defined BR in the automation rule (not just% rate = ...)
b) Integration of the cancellation rule into the actual automation
For example "Start Automation" if VALUE => nn and "Stop Automation" if VALUE <= nn

AdrianLxM, 2020-12-04:
It seems like you are using your own "loop" algorithm via automation on top of the existing loop algorithm.

doraberta, 2020-12-06:
That is certainly true. 
But doesn't everyone who use automation do that? 
... to get back to the target area faster; 
… to intervene when they want to exercise; 
… to intervene if their own diabetes is currently acting differently than AAPS is programmed, ...? 
(at least that's what the forums say) 
And that's what the automation was made for - wasn't it?

MilosKozak, 2020-12-07:
no. automation is there to give hints to existing alg. no to replace it

doraberta, 2020-12-08:
Hi Milos,
first: This is not the place to start discussions about the sense and purpose of automation, so I only make the following comments here:
1. It was a wish on my part, where I have explained the background.
2. Unfortunately, I cannot read the code and I therefore have to rely on explanations from others (as indicated above, the information from the forums indicate a completely different use). Is there somewhere for my understanding which hints the AAPS algorithm processes and how they influence it?

MilosKozak, 2020-12-08:
open this topic on fb page

closed 2020-12-08 by MilosKozak