nightscout/AndroidAPS #97: [2.7] connection to pump get lost (#72) didn't solve problem

thomasmenges, 2020-12-05:
Hello,
the closed Issue #72 doesn't solve the problem completely. I'm using the DEV-Version of AAPS to check if the closed issue is solved. Result: Better then in the master version (problem occurs not so often) but the problem still can appear.

What is the problem ?
During the day and also night it could happen, that APPS is waiting, waiting, waiting ..... for results of the pump after sending smb information or temp basal changes.

Bluetooth is still activ on my phone. Dexcom G6 delivers new BG values every 5 minutes correctly. After getting a new BG APPS tries to send change information to the pump but these information aren't correctly sent to teh pump. APPS is wainting for pump results in a loop and nothing happens.
The problem at night is, that this can happen for a long time and the high or low BG isn't changed anymore.

How can the problem be solved ? Easy by closing APPS and starting AAPS directly afterwards again. Now everything is OK up to the next problem that can happen hours or days later again.

Attached you can find some screenshot and the logfile.

Here you can see that since 4:00 am APPS isn't acting anymore
[image]

The latest sent change entry (TBR) to the pump was on 03:47 am
[image]

The latest SMB change entry was on 02:22 am
[image]

Here you can see, that BG values are delivered every 5 minutes correctly
[image]

And here you can see that AAPS is still waiting for a result from the pump. (I assume that no new change information was sent before although it should be)
[image]

And here is the logfile. Please use as time for the error log the range 09:01 am to 09:08
[AndroidAPS_LOG_1607155497361.log.zip](https://github.com/nightscout/AndroidAPS/files/5646923/AndroidAPS_LOG_1607155497361.log.zip)
 am

If you have questions please feel free to ask me. Thanks
Thomas

MilosKozak, 2020-12-07:
turn off sensitivity logging if you don't needed. it's too verbose

MilosKozak, 2020-12-07:
the problem happened before this log

thomasmenges, 2020-12-08:
@MilosKozak 
"the problem happened before this log". Yes that's the problem. Normally in the night you will not recognize that AAPS isn't working well and when you recognize it the issue was earlier than the logfile.

Yes you're right. Solving the problem by fighting against  the triggers is the best way but sometimes a medicine that covers a problem could also be helpful.

Normally I would assume that AAPs should know how long AAPS is waiting for a result of the pump (TBR and / or SMB). Is the time is linger than 360 sec why can't AAPS being restsrtet automatically and the problem is solved as soon as the bew BG is delivered?

By the way attached you can find a new log. Please look at the time range 16:12 to 16:17. Thanks in advance.

Thomas

[AndroidAPS_LOG_1607440623166.log.zip](https://github.com/nightscout/AndroidAPS/files/5660381/AndroidAPS_LOG_1607440623166.log.zip)

febe20, 2020-12-24:
@thomasmenges Hi Thomas, sadly i have the exact same problem. The only difference is that i use a Pixel 5 with Android 11. Did you find any solution or workaround yet? The 2.7.2 dev does not work for me either.

thomasmenges, 2020-12-26:
Hi @febe20,
good to know that not only me have the problems.

The problems aren't solved and occur 2 or 3 times a day. I'm really frustrated because the problem leads to bad BG values.

One thing can also be seen then the problem is alive.
A new connection to the DANA RS pump isnt't done anymore. Normaly you see on the pump that a bluetooth connection is on. But then the problem in AAPS occurs nothing is sent to the pump anymore.

And an other point I could see. Sometimes AAPS wants to send a temp basal change or a SMB to the pump and the result of the pump is 'false'. I do not know why the pump refuses to change temporarily the basal rate. 
Perhaps that can lead to the problem in this issue.

classchneider, 2021-01-03:
I am experiencing the same problem.

It is potentially unhealthy/dangerous when AndroidAPS for hours does not react on low/high BG, so a temporary fix as the one recommended by thomasmenges above should be implemented.

MilosKozak, 2021-01-03:
all i can say is that pump hangs during bolus and never disconnect from phone.
It's not clear if the problem is on RS or Android side but AAPS cannot resolve this situation.
Try another phone or pump
Anyway I added disconnect from Android side in this situation https://github.com/nightscout/AndroidAPS/commit/fd934fde2bd6dc22ce71b1e606ae9c8763f7576a

closed 2021-01-03 by MilosKozak

thomasmenges, 2021-01-06:
Hi @MilosKozak,
your mentioned changes in situation fd934fd solves the problem. So I'm using the DEV version and I'm lucky. Thanks

febe20, 2021-01-08:
@thomasmenges @MilosKozak Sadly its not working for me. Im sure now that everything has to do with my up-to-date Pixel 5 Android 11. All my Problems started after the new "Pixel Feature Drop"-Update in early dec 2020. damn

classchneider, 2021-01-08:
> Anyway I added disconnect from Android side in this situation fd934fd

Thanks for advice. I am looking into buying another phone.

By the way, I am using OmniPod Eros pump - your disconnect
correction mentioned above, is that only for danaPump?

In case your commit also applies for my pump, can you help med with a link
to the version to build?

Den søn. 3. jan. 2021 kl. 18.02 skrev Milos Kozak <notifications@github.com