nightscout/AndroidAPS #101: Make "INFO: Pump unreachable" notification actionable?

StephenBrown2, 2020-12-08:
As a user, I would like to be able to tap the "INFO: Pump Unreachable" notification that pops up and go directly to the "Medtronic" tab view.

I can see the `PumpUnreachable` `MedtronicNotificationType` here: https://github.com/nightscout/AndroidAPS/blob/bc43a926f9fce29eb3f3c262b4b6e955da088962/medtronic/src/main/java/info/nightscout/androidaps/plugins/pump/medtronic/defs/MedtronicNotificationType.java#L12 is used here: https://github.com/nightscout/AndroidAPS/blob/bc43a926f9fce29eb3f3c262b4b6e955da088962/medtronic/src/main/java/info/nightscout/androidaps/plugins/pump/medtronic/MedtronicPumpPlugin.java#L475 and here: https://github.com/nightscout/AndroidAPS/blob/bc43a926f9fce29eb3f3c262b4b6e955da088962/medtronic/src/main/java/info/nightscout/androidaps/plugins/pump/medtronic/MedtronicPumpPlugin.java#L570, but not being an app developer, I'm not sure how to add an intent to the dismiss/tap action. Possibly a check in https://github.com/nightscout/AndroidAPS/blob/bc43a926f9fce29eb3f3c262b4b6e955da088962/medtronic/src/main/java/info/nightscout/androidaps/plugins/pump/medtronic/util/MedtronicUtil.java#L235-L237?

andyrozman, 2020-12-22:
Notifications are just that... Notifications. They give you information, and then you can go and do something. 

Pump Unrechable notification is displayed when command can't be sent to pump.
 
Most of commands, will be retried automatically in 5 minutes (when next CGMS data comes in)...

ArthurusDent, 2020-12-22:
I've also wondered why I cannot tap on the notification and get at least straight into AAPS. That kind of breaks the typical Android behaviour. Getting straight into the Medtronic tab would be even better.

StephenBrown2, 2021-01-04:
Related, perhaps the notification can also be automatically dismissed when the pump is reachable again, similar to how https://github.com/nightscout/AndroidAPS/pull/118 added automatic dismissal of some notifications.

Alamo04, 2021-01-06:
@StephenBrown2 
I want to know that there was a notification,  if I missed one.

StephenBrown2, 2021-01-07:
So, funny enough, I've updated to 2.8 on dev, and now I'm not actually getting an Android notification, just the alarm sound and the in-app banner. Thankfully I know what that alarm sound means, but I could imagine a new user being completely baffled as to why their phone is making noise...

Harrypatz, 2021-01-13:
Is with me also so, was already at 2.7 and 2.8 and also at 2.8.1, that my phone starts to beep without what is displayed. Only by the sound I know that the pump can not be reached. Can I then only end in which I click in settings the app AAPS end force.

MilosKozak, 2021-01-14:
attach logs with time when this happen again

StephenBrown2, 2021-01-15:
Logs sent via email

Harrypatz, 2021-01-15:
[AndroidAPS_LOG_1610726020244.log.zip](https://github.com/nightscout/AndroidAPS/files/5821588/AndroidAPS_LOG_1610726020244.log.zip)

@MilosKozak 17:00

AdrianLxM, 2021-01-23:
> So, funny enough, I've updated to 2.8 on dev, and now I'm not actually getting an Android notification, just the alarm sound and the in-app banner. Thankfully I know what that alarm sound means, but I could imagine a new user being completely baffled as to why their phone is making noise...

If you dismiss a notification multiple times (3 times, I believe) Android now asks you if you want to disable this type of notifications. That's easy to click by chance/mistake. Can you please check in the App-Info if you have all notification channels for AAPS enabled?

Harrypatz, 2021-01-23:
@AdrianLxM 
Hello, yes it is all on, sometimes a notification window comes up where I can click ok. It is sometimes so and sometimes so

AdrianLxM, 2021-01-23:
@Harrypatz I believe we are mixing two different issues here. If you have logs for the situation you describe happening with 2.8.2, could you please open a separate issue. Then it's easier to directly ask questions regarding that topic. Thanks

AdrianLxM, 2021-01-23:
Thanks @StephenBrown2 for the feature request. It was not much work and will bring a lot of value!
https://github.com/nightscout/AndroidAPS/pull/258

StephenBrown2, 2021-01-23:
> If you dismiss a notification multiple times (3 times, I believe) Android now asks you if you want to disable this type of notifications. That's easy to click by chance/mistake. Can you please check [...]

Yes, same thing, but Adrian says he fixed the original issue, so I'll update to latest dev and see if I get notifications again.
[image]

closed 2021-01-24 by MilosKozak

StephenBrown2, 2021-01-24:
Noting the settings changed in the other PR, it seems I did not have "Use system notifications for alerts and notifications" enabled. Not sure if/how that changed, as the default has been false for over a year... @Harrypatz check your notification settings.