nightscout/AndroidAPS #106: direction arrow misplaced

guydavies, 2020-12-09:
In the 2.7.2 Dev build bf0dc1cb9 the direction arrow seems to be misplaced. 

I'm running on a Google Pixel4a

You can see the issue in the photo at the link below
[https://photos.app.goo.gl/YFG9LcSfiTfnfjDV7](url)

guydavies, 2020-12-10:
URL above should be [https://photos.app.goo.gl/YFG9LcSfiTfnfjDV7](https://photos.app.goo.gl/YFG9LcSfiTfnfjDV7).  Trying to do it from the phone didn't work out as I expected.

I'm seeing the same issue still with the very latest dev build 3f64dff01 built this afternoon.

StephenBrown2, 2020-12-11:
I have also seen an issue where the direction arrow gets updated from the bg source, then resets to horizontal shortly thereafter. Not sure if it's related, I'll try to capture logs next time it happens.

Alamo04, 2020-12-12:
Confirm guydavies issue.
Same on Pixel 5.

In the latest dev, it seems to be in place again, but this version can't be created,.
Time Ago position is now under the arrow and there are a few errors about time etc. in the building prozess.

guydavies, 2020-12-12:
Fixed in the latest Dev build.

closed 2020-12-13 by MilosKozak