nightscout/AndroidAPS #114: AndroidAPS splash logo

JimZam, 2020-12-12:
Good evening. I've a new Xiaomi phone and AndroidAPS splash screen appears randomly when the app  sends a new notification (I can see it in Notification Registry). Not everytime, just randomly, the splash screen appears and AndroidAPS comes to the foregroud. Any idea how avoid this behaviour?
Thank you and regards

MilosKozak, 2020-12-13:
what version? 
maybe AAPS crash and restart. check logs

JimZam, 2020-12-13:
Good evening. Fu****g Android Ongoing Notifications... My old phone had Android 7 and everything worked fine out of the box. I have been two days trying to avoid lock screen waking up every 5 minutes, stopping the phone playing a sound every 5 minutes while alarms in AAPS keep on working, preventing random AAPS splash screen, making xdrip alarms and notifications work as expected... I think finally everything works (I cross my fingers)
I'll check logs just in case but splash screen stopped appearing when I disabled AAPS "Show pop-up windows while running in the backgroud", maybe it could help someone in the future.
Thank you for your help and of couse thank you so much for this great, GREAT app.
Kind regards from Spain

closed 2020-12-13 by JimZam