nightscout/AndroidAPS #122: no was to get watch face installed through playstore-apps on phone...etc...

CaroGo, 2020-12-17:
Dana RS, AAPS dev from 17.12. , poco f1, fossil sport 

since dev version the aaps watch face does not get installed...if you go to playstore - app on your phone ...no aaps is displayed! no way to install! 
Building aaps.apk  is no problem... the wear.apk is there, it is not empty...everything seems correct..

guydavies, 2020-12-17:
As I said, the only way to install is using adb install -r
wear-full-release.apk from your computer with the watch on the USB charger
and debugging enabled on the watch.

CaroGo, 2020-12-17:
ok, but why is that necessary now? it was never before??
is it wanted to be that way? It will be difficult for most users...

osodebailar, 2020-12-17:
And how can we solve the problem ? Installation of wear.app with adb is not a practicable solution for a normal user .

guydavies, 2020-12-17:
I have a suspicion that it may have happened for me when I either updated
my old phone to Android 10 or got the new Pixel 4a which is Android 11. So
there may be new rules that prevent side loaded apps from uploading their
wear apps to a connected watch. Only apps from the Play store will work.
That's my suspicion.

CaroGo, 2020-12-17:
I didn't change anything, android 9.
I was able to install aaps with all watchfaces just as always. But now in dev it is not possible. So anything must have changed there. I can still install everything using master version...

guydavies, 2020-12-17:
OK. Must have been a coincidence for me then.

jotomo, 2020-12-17:
@CaroGo Make sure you're building the release variant (hover at bottom-left, select _Build variant_ entry in list).

rICTx-T1D, 2020-12-18:
@CaroGo  
I only know the way of sideloading too. The play store support only official apks published over google.
The fossil gen 4 and gen 5 doesn't have usb, but you can activate debugging over wifi. Then you can use the ip address to connect directly from computer to watch and useing abd-tool. Only thinking that  same key for codesigning is required, because connect between watch and phone is failed.

winni67, 2020-12-22:
It seems to be a problem in dev only. The AAPS 2.7 master works for me. My brother had problems like CaroGo on dev versions. I use Android10, AAPS 2.7 master with Fossil G5. No adb was necessary. Update to dev works for me, but I couldn’t test the watchface installation again. I still works for me.

Rig22, 2020-12-30:
Same problem, as @CaroGo  describes. The new watchface DigitalStyle01 @rICTx-T1D doesn't appear on my watch after install. I'm using 2.7.3 -dev.

rICTx-T1D, 2020-12-30:
@Rig22 
The problem isn't in watchface. When you uninstall AAPS from watch and will install again, the play store don't provide it under smartphone installed apps. When you take the side loading methode over dev-mode in WearOS, all working fine. I think it is possible that an update of wearos-smartphone-app could be make trouble with unofficial apps.

Rig22, 2020-12-31:
@rICTx-T1D 

I had uninstalled AAPS from the phone and the watch. During installation, AAPS reappeared in the Play Store on the watch and could be installed from there. The only thing missing was your beautiful watchface.

rICTx-T1D, 2021-01-01:
@Rig22  Iam really confused about this problem. On my Fossil 5 Gen, the play store don't show anythink from AAPS, so I install the wear.apk by abd through wifi-dev mode to watch.

rICTx-T1D, 2021-01-02:
The manual has more information to install wear.apk on your watch... See https://androidaps.readthedocs.io/en/latest/EN/Configuration/Watchfaces.html#building-wear-os-version-of-aaps

I hope this help anyone to deploy the watch-app, but the reasons for the problem is open and shouldt be explained.

MilosKozak, 2021-01-03:
test dev branch now
APK is smaller because of vector icons

CaroGo, 2021-01-03:
Sorry, in my case it did not work with actual dev branch! (2.8.0.1-dev build 2621045d6-2021.01.03-19:58) 😐

rICTx-T1D, 2021-01-08:
Apk build process will be restored with #198. Many Thanks!

CaroGo, 2021-01-08:
Cool! Funktioniert wieder! DANKE!

Rig22, 2021-01-09:
Danke @AdrianLxM  für den Fix und @rICTx-T1D für das schöne Watchface. Ist mein neues Standard Watchface. 👍

rICTx-T1D, 2021-01-09:
Now we need a hotfix for the last stable, so many users can use it :)... @AdrianLxM is a hot fix planed with this and many other fixes, or must we wait for next stable in few months?

closed 2021-01-10 by MilosKozak

winni67, 2021-01-18:
I have installed AAPS 2.8.1.1 master over the 2.8.0-dev, but there is no new AAPS watcheface installed on my Fossil G5 smartwatch. The old AAPSv2 watchface is there, after first installing the Fossil G5 smartwatch with AAPS 2.7 master. How can I further investigate, what‘s wrong?