nightscout/AndroidAPS #130: [DEV from 18.12.2020] RileyLink changes causes in many RL connection lost with no reconnect

Alamo04, 2020-12-21:
With the dev version, from 18.12.2020, there are changes done with RileyLink connection.

I think the changes are for Omnipod, but with this changes i have a lot connection lost to my Medtronic Rileylink 2.2. It doesn't reconnect automaticly, you have to close and restart the Bluetooth connection and/or close and restart AAPS to get the RL to connect again.

Last happen this night from 0:00 to 0:55, see log.
After this, I going back to the previous version.

[AndroidAPS_LOG_1608527696668.log.zip](https://github.com/nightscout/AndroidAPS/files/5726045/AndroidAPS_LOG_1608527696668.log.zip)

bartsopers, 2020-12-22:
I'm not convinced that this is caused by the recent minor changes we did to the RL framework for Omnipod. Is anyone else experiencing these issues? Do they persist or was it maybe just coincidental?

And just FYI, to answer your last question: the Omnipod driver has been developed in a separate project for about a year before we merged any code to the main AAPS repo. Since the Omnipod driver is in a testing phase and the code has been integrated into the main AAPS repo, the RL framework has been shared between Omnipod and Medtronic in order to prevent duplicate code.

ktomy, 2020-12-22:
Hi @Alamo04,
Frst of all, please consider what @bartsopers told, Omnipod driver was developed in a separate (from AAPS) project (together with Medtronic driver). It was just a matter of chance that Medtronic was integrated earlier (Andy could work on it more and the protocol is simpler).
The change we are talking about is valid for both Medtronic and Omnipod, as both use Riley(Ema/Orange/Loop)Link and RileyLink battery level is a valuable piece of information, which you want to have regardless of the pump you use. Again, it is just a matter of chance it was first available for Omnipod.
About the connection issue. In order to get the battery level AAPS does an additional call to the RL to get its battery level. To have this value in real time, this call is made after each connection to the pump. It may happen that if you have an unstable Bluetooth connection, giving the number of Bluetooth calls grow, the changes of connection failures grows as well. We have tested the feature on 5 different RileyLinks from different manufacturers, on different phones, and it worked flawlessly, so the feature was included in the development branch.
You are the first person who reports issues. It might happen that it is your particular setup, the RileyLink model or lot, the phone model or lot or the environment factors which negatively influenced the connection. Finding such kind of issues is the meaning of the development branch, so I would like to thank you for your report and would like to assure you we treat we will treat it with all the seriousness.
In order to better understand the context, would you mine to answer a few additional questions? It is important for us to know how would you appreciate the connection stability over time. How often does it happen for you to loose connection between the phone and the RileyLink? What do you usually do to restore the lost connection? How often does it happen that you loose communication link between the RileyLink and your pump? How often do you restart your phone?
Answers to these questions will really help us to address the issues as fast as possible. 
Besides that, we will work on a change which will prevent additional connections for getting the battery level if the RileyLink battery level indication feature is disabled.

Thank you for your understanding.

Alamo04, 2020-12-22:
@ktomy 

Thanks for your reply, also to you @bartsopers .

**Your questions:**

**It is important for us to know how would you appreciate the connection stability over time.** 
Before this update, the connection stability over time was great, no problems with RileyLink.
Sometimes connection to pump lost, but it reconnects automatically.

**How often does it happen for you to loose connection between the phone and the RileyLink?** 
- After different numbers of hours on the Poco F2 Pro, without reconnect and restart phone needed (send the phone back)
- Verry rare with my old Samsung S7. 
- Never with my Google Pixel 5 (until this update comes up)
  Normaly it happened only if the battery goes low... after 60 hours. 
  It should not happen, because I charge my RileyLink every 48 hours.

**What do you usually do to restore the lost connection?** 
Usually I stopp and restart Bluetooth. If that doesn't work, I close AAPS and restart it. Normaly it is fixed then.
Long time ago, it was needed to do a complete restart of the phone., 

**How often does it happen that you loose communication link between the RileyLink and your pump?** 
Sometimes and then perhaps 1 times a day, sometimes 2-3 times a day. It was normaly fixed with the next connection then.

**How often do you restart your phone?**
Only if I become problems with AAPS or XDrip, or if an phone update comes up.

**But it can be, that this was the problem. In the history, with a new experimental Medtronic AAPS version, it was needed to do a complete restart, sometimes to do a complete clean install with restart. Not need that for a long time now, but will try it after work tomorrow and give you feedback.**

andyrozman, 2020-12-22:
Is is really wise to check RL battery status on every call? In Medtronic driver I do some calls just like every 60/120 minutes (dependening on previous value)?

Alamo04, 2020-12-23:
@bartsopers and @ktomy 
I installed the build 691a867d8-2020.12.19 yesterday 23:56 again.
This time I restart the phone after installation. 

**Until now no connection lost comes up, so I think the restart of the phone has fixed it.
Will close this issue in a few hours, after this version runs 24 hours.**

**But another question:**
If you check the RL batterie so many times, did that not affect the runtime of the RileyLink ?

Alamo04, 2020-12-23:
After 23 hours with no problems this issue can be closed.

Restart phone has fixed this issue and the version is working good so far.

PR #132 (2020-12-23): Omnipod UI improvement & Revert RL changes for Medtronic
Closes #130
closed 2020-12-23 by Alamo04