nightscout/AndroidAPS #136: [2.7.2-dev] Graph shows temporary target of 0 for a few minutes

ArthurusDent, 2020-12-23:
While I had set an eating soon temporary target of 80, the target was changed to 0 for a few minutes. It's probably just a glitch because I couldn't find anything in the logs but I thought, one of the devs should have a look.

Screenshot:
[image]

Logs:
timeframe: 
01:02:11.543 [main] D/CORE: [OverviewMenus.onContextItemSelected():365]: USER ENTRY: TEMP TARGET EATING SOON
...
01:21:40.026 [main] D/CORE: [OverviewMenus.onContextItemSelected():408]: USER ENTRY: TEMP TARGET CANCEL

[AndroidAPS._2020-12-23_00-00-04_.zip](https://github.com/nightscout/AndroidAPS/files/5737479/AndroidAPS._2020-12-23_00-00-04_.zip)

System:
AndroidAPS 2.7.2-dev 6cb62412c
Medtronic 554
Dexcom G6
xDrip+
Oneplus 3T
Lineage 17.1

MilosKozak, 2021-01-03:
where is the zero?

ArthurusDent, 2021-01-03:
I've marked it:
[image]

MilosKozak, 2021-01-10:
this is caused by worng NS synchronization. turn on NS upload only

closed 2021-01-10 by MilosKozak

ArthurusDent, 2021-01-10:
I had already turned on "NS upload only" at the time this happened.