nightscout/AndroidAPS #138: Medtronic driver does not always retrieve the correct TBR

brmcc, 2020-12-24:
When AAPS sets temp basal to a higher rate, don't know the exact number yet, but it's safe to say 5 or hugher, this works as expected and the Medtronic tab will even say the correct rate until the next time it refreshes. Then it says around a third of the rate  even though the pump's screen verifies it as being what is set. 

In the log attached, I purposely ran my BG up and AAPS set a TBR at 10u/hr.  The status of the pump in the Medtronic tab and the main screen shows 3.60u/hr after the next poll of the pump.  At that time the pump display itself still shows 10u/hr on the status screen.

The problem is easy to reproduce and happens reliably with any higher TBR.
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/5740167/AndroidAPS.log)

I've updated my APK to the nightscout/AndroidAPS master branch.
Build bc43a926f-2020.12.24-10:45

andyrozman, 2021-01-03:
@brmcc You mean its showing 2.6 instead of 9, right?

brmcc, 2021-01-03:
Unfortunately my 2 week old RileyLink completely died almost exactly when you wrote this.  Since I no longer have one to test with, and this issue has never affected anyone else, I feel that we can close this.  I'll look for another solution for AAPS in the future that doesn't involve a RileyLink.  Thanks for your time looking into this!

andyrozman, 2021-01-03:
@brmcc While you are the only person reporting this problem, we do have problem in code... If amount of basal is too high, then value is not displayed correctly. So it needs to be fixed. 
About RL, talk to Jeremy (the guy sending RL from getrileylink.org) maybe he can help you.

brmcc, 2021-01-03:
Thanks for your willingness to help!  I went ahead and pulled my rileylink from the trash and figured out the battery and charging circuit is bad.  I got it going temporarily, and can in the future if needed to test if needed, and set the temp basal rate manually on the pump screen, disabled the loop, then clicked refresh on the AAPS medtronic tab so it would retrieve the TBR.  It seems maybe only 4 bits of data are read? It goes essentially to 64 then repeats itself.  I can include logs if needed. 

Here's my results:

0u/h - 6.30 u/h is correct
actual setting - displayed setting
-------------------------------------------
6.30 - 6.30
6.35 - 0.00
6.4 - 0.00
6.5 - 0.10
7 - 0.60
7.5 - 1.10
8 - 1.6
8.5 - 2.1
9 - 2.6
9.5 - 3.1
10 - 3.6
12.5 -  6.10
12.7 - 6.3
12.75 - 6.35
12.8 - 0.00
13 - 0.20

andyrozman, 2021-03-07:
https://github.com/nightscout/AndroidAPS/pull/400
PR for this problem

closed 2021-03-07 by MilosKozak