nightscout/AndroidAPS #143: [2.7] Using xDrip as source shows 0 as delta

nshtg, 2020-12-25:
When using xDrip as source (with G6) the delta in homescreen is (almost always) 0 for all three deltas.
When using the patched dexcom app as source this problem does not occur.

OpenAPS sees the real delta. Screenthots and partial log (for the affected tme +/- 5min) attached as zip.

[data.zip](https://github.com/nightscout/AndroidAPS/files/5742468/data.zip)

guydavies, 2020-12-26:
It sounds like you're getting the value from Xdrip, which is sent to AAPS
and to NS. Then you're getting the same value back from NS. Turn off all
reading from NS (NS uploads only). That should fix the problem.

nshtg, 2020-12-26:
@guydavies 
Did not fix the bug. Right after the reading comes in the homescreen shows the correct delta. But when I switch to a different app and come back its all on 0 again.
Right now using the patched dexcom app => AAPS => xDrip broadcast => xDrip uploads to NS

nshtg, 2020-12-26:
The bug is still existing in the dev branch (build from 3c32583d8c21b5726000a0a454c04b3c0a31c36d).

nshtg, 2020-12-26:
I made a recording to show what happens. BG gets updated at 14 seconds.

https://user-images.githubusercontent.com/5219415/103152211-ff0a0000-4785-11eb-878e-8ac0cfa4ce12.mp4

guydavies, 2020-12-26:
AAPS is likely uploading what it gets from Dexcom patched app as well as
Xdrip doing it. Stop Xdrip from uploading. It should only be there to
perform the alarms and presentation of data if you're using the Dexcom
patched app.

nshtg, 2020-12-26:
Disabling all inter-app communication + upload settings in xdrip and then **re-enabling BG upload in AAPS** did the trick.
I still think this qualifies as a bug, workaround or not!

guydavies, 2020-12-26:
I disagree. It works exactly as described in the documentation. Xdrip and
the patched Dexcom app are mostly redundant with each other. You're using
two apps to upload to NS and so as soon as the second uploads, it's a 0
difference from the first. Just stop Xdrip from uploading anything. If you
don't use it specifically for alarms or its presentation then you don't
need Xdrip at all.

nshtg, 2020-12-26:
You could argue like that for the direct delta. But what about the 15m and 45m delta? Those should not be set to 0 in any case.

guydavies, 2020-12-26:
I agree. If that happens it's a bug (or at least poor input checking) but
the incorrect duplicate input data would likely make the results wrong
anyway. If you send twice the data in the same time period, it'll skew the
results. I've never had an issue since I fixed the duplicate inputs when I
did the same as you. Fix the inputs and the outputs will be valid. If you
keep the inputs invalid then the output will be garbage irrespective of
whether or not it looks reasonable.

MilosKozak, 2021-01-03:
you setup DEX -> AAPS -> XDRIP - > NS -> APPS
so the system do exactly what you did setup. in the middle of chain data is changed by xdrip thus it's not deduplicated on AAPS side => deltas on 2 the same values are zero
it's not a bug on AAPS side, you did wrong setup outside (and/or inside) AAPS

closed 2021-01-03 by MilosKozak