nightscout/AndroidAPS #156: Profile helper text button text dark on dark theme

guydavies, 2020-12-30:
Transferred from MilosKozak/AndroidAPS at request of @andyrozman

When using the profile helper in the current Dev build and every time I've previously tested, the button at the top of the screen shows black text on a dark background so it is barely visible.

I suspect this is just not adjusting at all to the dark mode.

https://photos.app.goo.gl/eNxD8nNpie5RxY1e9

osodebailar, 2020-12-30:
test it on 2.7.3-dev and button show text and icon correct
[image]
[image]

Philoul, 2020-12-30:
The issue on this layout (minor) is for me the + and - buttons (black square around them and not centered vertically for "+" char)

guydavies, 2020-12-30:
Agreed. My issue does now seem fixed but I also see the minor issue
described by Philoul. It's quite irritating once pointed out to me 😉

Philoul, 2020-12-31:
It's link to ProfileHelperAppTheme style, but I don't know how to fix it.
I tried to use default AppTheme style (but AAPS crash), and I don't find which properties and setting to add/modify in ProfileHelperAppTheme to fix it.

rICTx-T1D, 2021-01-04:
> It's link to ProfileHelperAppTheme style, but I don't know how to fix it.
I tried it too and I don't solved it :(

But the vertical-alignment of labels are improved.

AdrianLxM, 2021-01-24:
Issue behind this: If the theme is a Material theme, the general Button will become a MaterialButton.
On a `MaterialButton` the `background` attribute alone will not work. `app:backgroundTint` should do it.

AdrianLxM, 2021-01-24:
:) Thanks @guydavies for taking this over. Fixed, I hope.

closed 2021-01-25 by AdrianLxM