nightscout/AndroidAPS #157: Virtual pump request (template)

Philoul, 2020-12-30:
PUMP TYPE:
Bolus step:
Extended bolus step:
Extended bolus duration step:
Extended bolus max duration:

Temp basal style: (percent or absolute)
Temp basal step:
Temp basal duration step:
Temp basal max value:
Temp basal max duration:

Base basal min value:
Base basal step:

rICTx-T1D, 2020-12-30:
More features requests:
* allow to switch of animation and waiting time during virtual isulin will be inject.
* define the volume of insulin cartridge and calculate the rest of amount
* make it possible to switch of temporary basal rate function (ICT users)

Philoul, 2020-12-30:
@rICTx-T1D not the purpose of this Issue.

closed 2020-12-30 by Philoul

andyrozman, 2020-12-31:
Just to ask what was purpose of this ticket? Did you want to have new virtual pump added?

Philoul, 2020-12-31:
It's just the template for new requests in future (there is a link in doc for new requests of virtual pumps, it was associated to an old ticket in Milos repository)
=> So I opened this ticket here and I replaced in doc https://github.com/MilosKozak/AndroidAPS/issues/863 by  https://github.com/nightscout/AndroidAPS/issues/157