nightscout/AndroidAPS #171: [2.8.0] Objective 3 not shown as solved, double entries with different results

winni67, 2021-01-02:
[image]
[image]
[image]
[image]

winni67, 2021-01-02:
This error was not found in the 2.7.x–dev version. Perhaps these changes causes the error
https://github.com/nightscout/AndroidAPS/commit/6ed733f8cf8fb423afda38147d10216d9d8953e4

guydavies, 2021-01-02:
> 6ed733f

I see the same error. I have completed all the objectives under 2.7.x-dev
so it's not an issue of switching between an older version and a version
with the new objectives. Interestingly, it has no effect on automation,
which was always prevented when objectives weren't previously completed.

galloglass, 2021-01-02:
I have the same issue, I can't see anything in the release notes to say we must complete obj3. All other objs complete. Do we need to complete this new version of obj3?

guydavies, 2021-01-02:
Looks like we need to complete these objectives. Working through them. Was
confused by the wording still on three of them (English is my mother tongue
so that should not be an issue 😉). If I have more issues with those
questions, I'll raise issues here.

galloglass, 2021-01-02:
> Looks like we need to complete these objectives. Working through them. Was confused by the wording still on three of them (English is my mother tongue so that should not be an issue 😉). If I have more issues with those questions, I'll raise issues here.
> […](#)

Ah Guy, say it ain't so.... I'm in a cold sweat thinking of going back at obj3. May the force be with us lol

winni67, 2021-01-02:
Answering all the new or changed questions solved the problem for me. But I don‘t understand, why completed objectives are poping up as uncompleted after an software update. That is confusing me.

MilosKozak, 2021-01-03:
new questions were added. they must be answered
only changed questions remains marked completed

closed 2021-01-03 by MilosKozak