nightscout/AndroidAPS #172: [2.8.0] unchanged CGM values error with patched Dexcom app

the-swish, 2021-01-02:
Release notes states: removed "Unchanged CGM values" behavior in SMB for dexcom native app

I get still the message with patched Dexcom app.

`isSaveCgmSource` is true in the logs but still the message _"Error: CGM data is unchanged for the past ~45m."_ appears.

```
16:24:58.757 [RxCachedThreadScheduler-1] D/APS: [DetermineBasalAdapterSMBJS.invoke():112]: isSaveCgmSource: true
16:24:58.764 [RxCachedThreadScheduler-1] D/APS: [DetermineBasalAdapterSMBJS.invoke():165]: Result: {"reason":"Error: CGM data is unchanged for the past ~45m. Replacing high temp basal of 3.125 with neutral temp of 1.25","deliverAt":"2021-01-02T15:24:58.753Z","temp":"absolute","duration":30,"rate":1.25}
```

Looks like `isSaveCgmSource` is not true in the `determine_basal` function.
It also looks like the current implementation will only dismiss the message _"Error: CGM data is unchanged for the past ~45m"_ if `isSaveCgmSource` is true, but will still change the results if the CGM data is unchanged, because `isSaveCgmSource` is missing in the second `if`.

rICTx-T1D, 2021-01-02:
What's your question or problem with this topic?

the-swish, 2021-01-02:
3442e29e5c95a50edf85f838b922bb6294df7091 should allow flat BGs with Dexcom, I think it was a change for issue https://github.com/nightscout/AndroidAPS/issues/35
It looks not to work like intended.
Because I get still the message _Error: CGM data is unchanged for the past ~45m_

And if this will work the only thing it does is to remove the message _Error: CGM data is unchanged for the past ~45m_ but the change of the high/neutral basal still happens and no prediction line will be shown because it is missing `&& !isSaveCgmSource` in the second compare of `determine_basal`

MilosKozak, 2021-01-03:
fixed by https://github.com/nightscout/AndroidAPS/commit/34435d46847fd2f4783a97a5a90de5ba7bb41f98
thanks, 2 bugs in a single commit. shame on me

closed 2021-01-03 by MilosKozak