nightscout/AndroidAPS #173: AAPS 2.8. not possible to enter / edit automation name

CaroGo, 2021-01-02:
Version 2.8., poco f1 android 9, DanaRS, 
In this new update it is not possible to name a new automation or edit the name of an already existing automation. You can enter the field, but cannot edit...

guydavies, 2021-01-02:
Did a bunch of new, incomplete questions appear in your objective 3. They
did for me. I haven't tried updating automation rules because I've also not
finished the questions which aren't clear to me still.

Rig22, 2021-01-02:
I can confirm the reported issue @CaroGo.

guydavies, 2021-01-02:
Yep. Me too. I have completed all the new questions and tried exiting and restarting AAPS just to be sure. Like the OP, I could add conditions and actions but the textbox for the title doesn't open a keyboard on screen.

MilosKozak, 2021-01-03:
fixed by https://github.com/nightscout/AndroidAPS/commit/d6f00e5b0d66c6ae12c69b73b400d38d450bf194

closed 2021-01-03 by MilosKozak