nightscout/AndroidAPS #182: [2.8] Setup Wizard fails to import configuration

ArthurusDent, 2021-01-04:
I switched phones yesterday and did a fresh install of AndroidAPS 2.8 but copied the config files, too. When the Setup Wizard offered to import my settings, I pressed the button to open the dialogue but nothing happened. Another person on Discord reported the same problem. The person and I had no problem importing the config by using the usual maintenance dialogue.

System:
AndroidAPS 2.8
Medtronic 554
Dexcom G6
xDrip+
Oneplus 3T
Lineage 17.1

MilosKozak, 2021-01-04:
on some phones it's necessary to restart aaps to apply storage permission

ArthurusDent, 2021-01-05:
I didn't restart AndroidAPS. I quit the wizard and went straight to the maintenance dialogue.

MilosKozak, 2021-01-05:
then restart of activity is enough

ArthurusDent, 2021-01-05:
Sorry, I'm not following. Could you repeat yourself in more layman terms?

MilosKozak, 2021-01-05:
switch between aaps screens

robertrub, 2021-01-06:
Had the same issue. Import settings while installing didn't import the saved settings. I'm not 100% sure that I restarted the app after the permissions but I think I did restart. 
I had to quit the creation of a new profile and imported through the classical maintenance menu and it worked.

closed 2021-01-24 by MilosKozak