nightscout/AndroidAPS #186: TDD in Statistics is misscalculated

cristinav74, 2021-01-05:
Build version: 2.8

[image]

The TDD for today is zero, which is incorrect. This results in the Average 8 days results being corrupted.
I believe the Average results should exclude current day.

rICTx-T1D, 2021-01-05:
In my 2.8.0 version, the stat is ending yesterday and the average is over 7 days.

Do you have change time/date or timezone last 7 days?

cristinav74, 2021-01-05:
No, but my phone had a security update last night.

New problem: after restarting my phone, I have completely different results in Statistics (obviously corrupted)
[image]

rICTx-T1D, 2021-01-05:
This don't look nice. :-)
* Store all settings by export  on maintenance menu. 
* Sync all values with nightscout. 
* reset the stat's
* reset the database in maintenance menu
* check if is solved, otherwise reinstall AAPS

cristinav74, 2021-01-05:
Tried all the options, problem still not fixed but thank you for your suggestions.
As a side detail, I migrated to 2.8.0 from 2.7.0 on Janury 3rd, which is the last day when the numbers are correct.
[image]

rICTx-T1D, 2021-01-05:
Has anyone other user the same problem? I can't reproduce it.

cristinav74, 2021-01-05:
[image]

It looks like the pre-migration data cannot be loaded. No TIR and average for the days before installing 2.8.

rICTx-T1D, 2021-01-05:
The the problem will be solved when time is gone.... Wait some days.

cristinav74, 2021-01-05:
Definitely, it is a minor problem and should not affect future data (I hope so, I'll monitor my Stats).

rICTx-T1D, 2021-01-05:
Can you check your stats in 7 or more days (calculated from update-date) again? When all is fine, you can close this issue, otherwise we need a new run to solve it again.

cristinav74, 2021-01-05:
Sure, I'll keep this issue updated

CaroGo, 2021-01-06:
I can't reproduce it either... just 7 days shown as it should be. Dana RS,

cristinav74, 2021-01-11:
The Statistics are correctly calculated now that all of them have been logged under 2.8 version.

closed 2021-01-11 by cristinav74