nightscout/AndroidAPS #189: Be able to delay activity time

robertrub, 2021-01-06:
Suggestion:

When I enter carbs, I can say that it was x min earlier or will be x min later. That's great.

It would be useful to be able to input the delay for activity also. I started an activity and gorgot to input it v input activity with -x minutes (this will be just informational or may help the auto tune/adapt) or I'll start the activity in +x minutes (AAPS may inform the user the need to stop the basal and/or eat carbs).

guydavies, 2021-01-06:
Isn't activity purely a careportal note?  It has no actual bearing on the
behaviour of the algorithm AFAIK.

robertrub, 2021-01-06:
Yes and no ;) If you enter activity via _carbs_ menu (even without entering carbs) you can delay + or - (thus my request is no more justified ;)  ) and the target changes.
If you enter via the _activity button_ on page 2, you have no delay and no modification to the target... 
Strange that there are 2 different reactions to the same info input.

guydavies, 2021-01-06:
Do you mean starting the activity TT?  That's the only thing relative to
activity I see from the carbs button. Doesn't a TT only have relevance into
the future. What's happened in the past is done and changing the target for
that period won't change the behaviour of the algorithm AFAIK. I hope it
doesn't.

Adding carbs into the past makes sense because those carbs may still not be
decayed and so the remaining carbs will impact the algorithm in the future.

robertrub, 2021-01-06:
Ok, I get the logic of it. 
It would be nice (just for personal reference) to note that the activity started some time ago and I can note it in the carbs menu.
It is ok as it is and I'm closing this request :)

closed 2021-01-06 by robertrub