nightscout/AndroidAPS #193: [Improvements] Secondary graphs (use same scales if same units on the same graph) and replace Activity by -BGI

Philoul, 2021-01-06:
See below 2 suggestions concerning secondary graphs to open discussion:
1) When several curves with "same units" are in the same subGraph, it could be nice to calculate scale with all values and not separatly (for ex. if we put Absolute Insulin and Active Insulin on the same graph, then use the same scale for both curves) to be able to directly compare curves.
 
2) It would be nice (I think) if we replace for secondary graphs Insulin Activity (redondant with Activity available in main graph) by Blood Glucose Impact (-BGI to be exact)
-BGI is very close to Activity (these curves are proportional if you have only one ISF value in your profile), and we can directly compare -BGI values with deviations values (so curve compare only if same scale is used for both curves, see above1st suggestion)
=> If deviation = -BGI then BG value remains constant, 
=> If deviation > -BGI then BG increases,
=> If deviation < -BGI then BG decreases
I think BGI is better (especially if profile contains multiple ISF values) for analyzing the impact of insulin on BG variations

Just to illustrate the second proposition, I took a screenshot when the Activity curve was very close to "-BGI value" with the same scale than deviation:
On the left BG value is constant (deviations equals "-BGI"), then between red lines deviations are greater than -BGI (BG rise), and finally on the right BG is constant again...
[image]

Philoul, 2021-01-26:
@AdrianLxM 
I was waiting to have a feedback on this proposal before looking at the code.
I can try to look if I can do it if you think it's a good proposition. (I don't know yet if it's complicated or not...)

AdrianLxM, 2021-01-26:
@Philoul I didn't have the time to fully think that through.
Do Absolute insulin and "insulin activity" have the same unit?

BGI and delta definitely are comparable - change and expected change -> That would be great if they had the same scale if they are in the same graph!

Philoul, 2021-01-27:
> Do Absolute insulin and "insulin activity" have the same unit?

No, but Absolute Insulin and Insulin on board yes... (my bad, in french IOB is translated IA for Insuline Active, I updated my proposal to avoid misunderstanding, thanks)

Philoul, 2021-01-30:
@AdrianLxM what do you think of this proposal #276 ?
=> Should I change -BGI color? (identical to the activity's color in this first commit)

closed 2021-02-12 by Philoul