nightscout/AndroidAPS #195: AndroidAPS changes automatically the language in to English.

whynot2018, 2021-01-07:
Android settings are in German and Auto-rotate screen on. During the answering 3.Objectives, if you rotate the screen into landscape mode it changes automatically the language into English.

[image]
https://files.gitter.im/5779284ec2f0db084a2135e1/Bbin/Screen_Recording_20210107-020356_AndroidAPS.mp4

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

guydavies, 2021-01-07:
There should be an option to select the system language in AAPS. Make sure
to select the option that actually says "system language" rather than
choosing your language. That's the only way that things are guaranteed to
work entirely as expected.

whynot2018, 2021-01-07:
Yes, we need to change both Android settings and AAPS language into the same language.
Thanks.

closed 2021-01-07 by whynot2018