nightscout/AndroidAPS #199: Basal is On when 0 Basal is registered

smokkelaar, 2021-01-08:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

bartsopers, 2021-01-08:
This is once again a dangerous situation because of NS upload only was not enabled. @smokkelaar now set NS to upload only and it works now.
@MilosKozak we **really** need to fix this, either by completely disabling the option to download from NS, or by not allowing NS to change the truth (= create/overwrite treatments in the present).

bartsopers, 2021-01-08:
[image]
@MilosKozak you can see here that this treatment keeps being bounced back from NS. You can also see it in the attached logs.
[AndroidAPS._2021-01-07_23-55-51_.txt](https://github.com/nightscout/AndroidAPS/files/5784868/AndroidAPS._2021-01-07_23-55-51_.txt)

bartsopers, 2021-01-08:
Another option would be to allow to backfill from NS once on intial setup, when no data is present yet. Anyway in the current form it is dangerous. It frequently causes all kinds of problems and it should be removed ASAP.

MilosKozak, 2021-01-08:
this will be resolved by new NSClient

closed 2021-01-08 by MilosKozak

bartsopers, 2021-01-08:
We should at least show a proper warning under that setting in the mean time.

Rhabarbara745, 2021-01-08:
I agree with @bartsopers 
I've had situations where aaps administered way too much Insulin (**smbs given twice, the full amount of maxSMBBasalMinutes delivered every single time although there was already too much IOB, etc)** because NS sporadically corrupted treatments in aaps!

=> Downloading data from NS should be completely disabled until we have a safe solution  - my two cent

MilosKozak, 2021-01-08:
it's in wiki and default is upload only .... what can we do more?

MilosKozak, 2021-01-08:
and think it's wrong xdrip configuration in many cases

bartsopers, 2021-01-08:
@MilosKozak Add a text in the setting description? Something like "WARNING only disable this temporarily to backfill data from NS. Continuous download from NS may cause unexpected and unwanted behavior"

bartsopers, 2021-01-08:
And I agree with @Rhabarbara745. The only real right thing to do for now is to completely disable it.

MilosKozak, 2021-01-08:
for parents is still very usefull and if you configure everything carefully it works

bartsopers, 2021-01-08:
And if you don't configure it carefully, you can easily end up in life-threatening situations. That's unacceptable IMO.
Also, parents can use sms commands instead, right?

ZandMarc, 2021-01-08:
Agree with @bartsopers and @Rhabarbara745 that preferrably this option is disabled or least a very clear warning should be under it. In the Dutch Looped group I moderate I see multiple posts from people that had this option switched on unknowingly. And to be perfectly honest, even I - a seasoned user of OpenAPS and AndroidAPS (3,5 yrs) made a mistake once with it (resulting in too little insulin luckily, not way too much).

MilosKozak, 2021-01-08:
> And if you don't configure it carefully, you can easily end up in life-threatening situations. That's unacceptable IMO.
this is valid for 30% of settings

MilosKozak, 2021-01-08:
if I disable it just after next release there will new issue to put it back
we can added a warning if you have an idea how to do it

ToreBj, 2021-01-08:
> this will be resolved by new NSClient

When is this scheduled to happen?

MilosKozak, 2021-01-08:
some work has been done, but still not ready

bartsopers, 2021-01-08:
> > And if you don't configure it carefully, you can easily end up in life-threatening situations. That's unacceptable IMO.
> 
> this is valid for 30% of settings

The difference here is that for most settings, it's possible to know to consequences when you read the docs. It's impossible to know all the unwanted behavior caused by this feature.

I would be way happier with disabling it altogether, but otherwise, can we please include the warning in the next release? I'm away for the weekend and I didn't bring a computer so I can't create a PR myself unfortunately.

MilosKozak, 2021-01-08:
added more specific warning
https://github.com/nightscout/AndroidAPS/commit/c847a5a3407c38440e3ecdb784c8172132ffc5cb

smokkelaar, 2021-01-08:
Thanks all for regonizing this issue, and for the quick help for a quick fix for my problem.

Some notes from my side.
@MilosKozak I agree with you that 30%+ of the settings within Xdrip is lifethreatening when the setting is wrong. BUT, when you change settings you are aware what is happening and where to look. This issue can pop-up at a random time that makes it very difficult for the user to pinpoint the problem.
Also because it has direct influence on the BG, and afterwards it still has trouble to give the correct amount for the next DIA time, because the system doesn't know any more how much insuline is really given!.

This program must be programmed in a secure way, that means: NEVER trust the input unless.... so you must check if the values that are imported are valid. So in this issue there are already some examples from people that really use this function and generating "valid values"
1. other people (mostly parents) that can interfere through nightscout.
2. External sources for BG values.
3. An extra input for food.
4. and there will be more cases.

That also means that when 2  sources are fighting (giving conflicting data) we can see that and notify that. big difference between BG on the device and nightscout. more then 3 insuline adjustments within 20 minutes. or getting a Carb correcting within 5 minutes from 2 sources (local and nightscout).
We should be able to notivy the user that something strange is going on.

Secondly, we see that the omnipod reacts badly on some changes. see my example. In that case: disable it now!. you cannot trust the functionality even with a valid case from above.
if omnipod -> set setting on disabled.  you can even disable the function so you can't select it.

My short opion:
Never trust a source!.
Don't set this at closed without prove it is already solved and available at the other side. 
Fix it anyway, don't trust a other source!.

ways to fix it in my opinion:
1. Disable the function.
2. check for strange behaivor.
3. extent the function to which data is accepted (BG / kalibrations / Bolus / basal / carbs / profiles).

After I typed everything i saw that you added a specific warning...  👍

bartsopers, 2021-01-08:
@MilosKozak I saw that you added the warning, thanks. However I don't fully agree on the warning. It also causes problems when nothing is configured incorrectly. This GH issue that were commenting on here, is an example of that. This is NOT a PBCAK.

teleriddler, 2021-01-08:
I would also like to express support to disable NS sync until a permanent solution is in place, this is very dangerous and we have had several users already comment that unexpected lows have occurred due to this setting being enabled (which it should not be by default).   I understand the need for parents to give treatments to their children through NS, however SMS can do the same thing.  Coming over from IOS LOOP I know what a great feature OVERRIDES for NS  are for parents giving remote treatments.   

The big difference I see here is the same as stated above, most settings you can get a good idea of what will happen when it is turned on/off and it is easily to validate that change.   This issue differs as it happens in the background with no warning for the user.  We have enough settings and inputs to troubleshoot already when BG values fluctuate for no reason and this just makes it harder to diagnose.  Usually you look at your ISF, IC, Basal Rates, hardware connections, what was eaten, exercise and more in order to ascertain what is causing the issue.  However if you are relying on your treatments window to give you accurate information to make a decision/diagnose and those treatments are being overwritten with false values.....then everything else mentioned earlier does not matter because you are troubleshooting with incorrect values.  I would say this is similar to having a CGM reporting values 40 mg/dL or greater difference from actual blood samples.   Making dosages and giving treatments with that information is simply unsafe (hence why we don't allow Libre to participate in full SMB functionality).  

A possible solution would be to use the button to do a temporary backfill but I still think that leaves room for error as well.

+1 from me for feature disable until a permanent solution is in place and tested.

afreitag33, 2021-01-08:
+1 from me for feature disable as well
For me the issue is, as others have said, that it's a sneaky potential failure. Usually when I change settings, I check around to see if everything is working properly for a couple of hours and then move on with my day. But it's possible that NS Sync works great for those couple hours and then a sync breaks later, and I would not think to look back to settings changes right away. First I'd look at what happened immediately before that could be causing issues - and as teleriddler said, these values would be present but wrong, and not flagged in any way. That's problematic. I also like the temporary backfill idea, since it seems that's the primary use of syncing (since parents should be using SMS anyways, as it's safer).

ArthurusDent, 2021-01-08:
This feature should be disabled now. In its current form it's too dangerous.

bartsopers, 2021-01-08:
@MilosKozak can you please reconsider?

larsqvd, 2021-01-08:
I actually use the feature to enter my insulin injections when I sometimes take a short break between pumps. But the sneakiness that afreitag33 mentions is dangerous indeed as I don't check this frequently for errors. Potentially only enable on engineering mode for the moment for those who are really desperate to use it (despite the multiple warnings)?

eurenda, 2021-01-09:
I've been following this issue and even with the warning in [c845a5a](https://github.com/nightscout/AndroidAPS/commit/c847a5a3407c38440e3ecdb784c8172132ffc5cb) it proves dangerous and problematic to fully troubleshoot and  overwriting of treatments and urge to please reconsider disabling the _download treatment from NS option_ until a more permanent and safer fix is enabled

vanelsberg, 2021-01-09:
Fully agree with @bartsopers and other posters that this problem is potentially dangerous and hard is to recognize as the cause of problems. _Especially the fact that parents will use this for me only adds to the urgency to handle this efficiently!_

I can not see how setting defaults and/or an additional warning is really helpful here:
- Unaware of the problem, the majority of users will upgrade there existing installation or import settings, totally undoing any defaults or an additional warnings.
- One can not expect the user to fully understand the implications of enabling NS-sync while the problem exists.
- Most users will just accept the warning?

@MilosKozak Given the risks involved on major injurie,
until the problem is fixed in NSClient,
Please reconsider fully disabling NS-sync in settings (maybe only enable for engineering_mode?)

vanelsberg, 2021-01-09:
Suggestion:
If for technical reasons disabling the option from preferences is a problem, how about automatically disable NS-sync on AAPS startup when not in engineering mode. This would enable the user to backfill while getting the warning message?

dar-kod, 2021-01-11:
I also agree to disable this feature as long as it is not fully functional. It is not just dangerous for a child but for a parent who wants to help a child as a parent may get a sense of own guilt if something bad happens. That would then all together affect the extraordinary reputation for AndroidAPS, what should be avoided at all costs.

MilosKozak, 2021-01-12:
for 2.8.1
NS_UPLOAD_ONLY is forced to be true
change is allowed only in engineering mode
engineering mode has been reset for everyone

vanelsberg, 2021-01-12:
@MilosKozak Until the NSClient fix is available, imho a perfect solution! Thanks!

Rhabarbara745, 2021-01-13:
Thank you very much for rethinking about this issue and making APPS safer and less prone to errors! @MilosKozak

tynbendad, 2021-02-05:
can someone describe how to avoid the original issue if someone were to enable engineering mode and disable NS_UPLOAD_ONLY  ?  i don't understand how the description in this issue relates to NS or what xdrip settings were incorrect for original poster?