nightscout/AndroidAPS #205: Wrong Pod State when Pod connection failure

philou78, 2021-01-09:
With AndroidAPS 2.8.0 build on 2021/01/02, I got a connection failure with my Pod :

[image] [image]

I was a little bit confused with the Pod State = "Running". 
Pod State was instead: "Connection failed" or "Not available"

Is it possible to improve this ?

AdrianLxM, 2021-01-24:
@philou78 I'm not clear what was the expected and what the actual behaviour.

Even though the connection fails, the pod would still be running, wouldn't it?
Maybe the label can be clearer?

philou78, 2021-01-25:
@AdrianLxM : Actually no, I had to replace it.

On the Pod screen, there was no problem except an uncertain TBR, but the real state was that my POD was no longer working.

I reported this because it was an uncomfortable situation for me.

When the connection RL<->POD fails, can we change Pod State with " Pod Connection failed" ?

AdrianLxM, 2021-01-25:
@bartsopers what's your opinion on this?

bartsopers, 2021-01-26:
Like @AdrianLxM said, this field reflects the Pod state. There is a separate RileyLink state field in the Pod overview.

My advice would be to configure the Pump Unreachable Alarm threshold to a value that you feel comfortable with. You will then get an alarm when the communication fails somewhere.

philou78, 2021-01-26:
I certainly mis-described the situation and it will be easier to describe it with the attached logs:
Suddenly, AAPS could no longer decode the responses from the POD.
All of my attempts to manually refresh its state resulted in "MessageDecodingException" exceptions.
Despite this, the status of POD remained "Running".

In this (exceptional) case, the POD was no longer in the "Running" state. No ?

Philou

PS : @bartsopers, thanks for the tip.

[AndroidAPS._2021-01-09_10-42-50_.txt](https://github.com/nightscout/AndroidAPS/files/5876191/AndroidAPS._2021-01-09_10-42-50_.txt)

afreitag33, 2021-02-03:
Similar thing happened to me! Got a screamer, but AAPS said it was running. There were no error notifications, but in the pod history it said "Communication failed: failed to decode message from Pod". Logs attached, occurred at around 9:40ish pm. 
I've had loads of static issues, so it totally could have been that, but generally that gets an error notification or some recognition from AAPS, so this was different.

[AndroidAPS_LOG_1612320583507.log.zip](https://github.com/nightscout/AndroidAPS/files/5918949/AndroidAPS_LOG_1612320583507.log.zip)

MilosKozak, 2021-10-08:
@bartsopers is this resolveable?

closed 2021-10-11 by MilosKozak