nightscout/AndroidAPS #208: [2.8.0] rest of insulin will not decrease by using virtual pump

rICTx-T1D, 2021-01-09:
Hi community, 

Iam using virtual pump mode and detect in version 2.8.0 (master) that the amount of insulin will not smaller, when insulin was recorded in bonus page.

When I start a new cartridge,  I make a infusion-set change. Then amount is full and age will be displayed as new. After some days and injections later, age has forwarded but amount is unchanged. 

I think it is a bug. Maybe in a early version too.

MilosKozak, 2021-01-09:
do you have screenshots?

rICTx-T1D, 2021-01-09:
[image]

MilosKozak, 2021-01-10:
Please specify pump and upload logs

MilosKozak, 2021-01-10:
sorry i missed the point.
Virtual pump is not capable to track resorvoir level

closed 2021-01-10 by MilosKozak