nightscout/AndroidAPS #212: [2.8.0] AAPS Crashes 1-2 times a day

mtothearkus, 2021-01-10:
Hi,

since Updating AAPS from 2.7.0 to 2.8.0 I noticed, that it crashes once or twice a day.

I did not change anything on the system, just the App Update.

Android 9, Sony Xperia Compact XZ2
Dexcom G6 (patched APP) (sending through AAPS, which sends to xdrip+)
Accu-Chek Combo

the following happens: I notice that xdrip+ doesn't receive bg values anymore and I open AAPS on the phone. Than it is the same behavior like a phone restart. AAPS initializes again and the BOLUS, INSULIN and Calculator do not appear. The pump is again being initialized and after the connection worked, the three Buttons appear again.

In the logs there is nothing special I think. You see an hour of no entries, which shows that AAPS crashed/stopped nearly an hour before.

`11:14:22.165 [RxCachedThreadScheduler-37] D/GLUCOSE: [GlucoseStatus.getGlucoseStatusData():170]: Glucose: 157 mg/dl Noise: 0 Delta: -4 mg/dlShort avg. delta:  -3,22 mg/dl Long avg. delta: -1,03 mg/dl
11:14:22.165 [RxCachedThreadScheduler-37] D/AUTOMATION: [TriggerBg.shouldRun():62]: NOT ready for execution: Glukosewert ist gleich oder kleiner als 84 mg/dl
11:14:39.658 [RxCachedThreadScheduler-19] D/NSCLIENT: [NSClientPlugin.lambda$onStart$4$NSClientPlugin():168]: PING received
12:16:24.727 [main] D/PUMPBTCOMM: [RileyLinkBLE.onInit():263]: BT Adapter: android.bluetooth.BluetoothAdapter@9694bd
12:16:24.731 [main] D/CORE: [RFSpy.onInit():76]: RileyLinkServiceData:info.nightscout.androidaps.plugins.pump.common.hw.rileylink.service.RileyLinkServiceData@89d1ab2
12:16:24.756 [main] D/CORE: [MainApp.onCreate():68]: onCreate
12:16:24.781 [main] I/DATABASE: [DatabaseHelper.onCreate():136]: onCreate
12:16:24.854 [main] D/CORE: [MainApp.onCreate():86]: Version: 2.8.0
12:16:24.854 [main] D/CORE: [MainApp.onCreate():87]: BuildVersion: d40c5079d-2021.01.02-12:13
12:16:24.855 [main] D/CORE: [MainApp.onCreate():88]: Remote: https://github.com/nightscout/AndroidAPS
12:16:24.862 [main] D/CORE: [PluginBase.setPluginEnabled():79]: Starting: OpenAPS SMB
12:16:24.863 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():78]: Defaulting APSInterface
12:16:24.864 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisiblities():185]: Selected interface: OpenAPS SMB
12:16:24.864 [main] D/CORE: [PluginBase.setPluginEnabled():79]: Starting: Rapid-Acting Oref
12:16:24.864 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():89]: Defaulting InsulinInterface
12:16:24.865 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisiblities():185]: Selected interface: Rapid-Acting Oref
12:16:24.865 [main] D/CORE: [PluginBase.setPluginEnabled():79]: Starting: Sensitivität Oref1
12:16:24.865 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():99]: Defaulting SensitivityInterface
12:16:24.866 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisiblities():185]: Selected interface: Sensitivität Oref1
12:16:24.866 [main] D/CORE: [PluginBase.setPluginEnabled():79]: Starting: Lokales Profil
12:16:24.867 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():109]: Defaulting ProfileInterface
12:16:24.867 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisiblities():185]: Selected interface: Lokales Profil
12:16:24.868 [main] D/CORE: [PluginBase.setPluginEnabled():79]: Starting: gepatchte Dexcom App
12:16:24.868 [main] D/CONFIGBUILDER: [PluginStore.verifySelectionInCategories():119]: Defaulting BgInterface
12:16:24.868 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisiblities():185]: Selected interface: gepatchte Dexcom App
12:16:24.869 [main] D/CORE: [PluginBase.setPluginEnabled():79]: Starting: Virtuelle Pumpe
12:16:24.870 [main] D/PUMP: [VirtualPumpPlugin.refreshConfiguration():410]: Pump in configuration: 1, PumpType object: GenericAAPS
12:16:24.870 [main] D/PUMP: [VirtualPumpPlugin.refreshConfiguration():412]: New pump configuration found (GenericAAPS), changing from previous (null)`

Corner4566, 2021-01-10:
I had the same issue - before( with 2.7) i could choose in the energie menue that AndroidAps would be not touched with energie safety - now it is not chooseable

MilosKozak, 2021-01-10:
Sorry I didn't find any crash reports comming from XZ2 as well.
try different phone. I'm not able to help

closed 2021-01-10 by MilosKozak