nightscout/AndroidAPS #213: Dana RS not regularly reading reservoir

JohnKitching, 2021-01-10:
I noticed yesterday AAPS was reporting 40 units in the reservoir for several hours.  
On tapping the Bluetooth icon it dropped to  23. 
Today I took a couple of screenshots showing before and after tapping the Bluetooth icon. 
I'm on 2.8. I didn't notice this happen on 2.7 but I'm not certain it didn't. 
Will email logs now. 
[image]
[image]

JohnKitching, 2021-01-10:
Sorry, screenshots wrong way round.

MilosKozak, 2021-01-10:
there is not enough information in logs.
you turned on autosens logging which is producing so much data that it's effectively breaking find something else
open new issue when you have more useable logs

closed 2021-01-10 by MilosKozak