nightscout/AndroidAPS #217: [2.8] "History Browser" window does not display/refresh as requested

jbuckhi, 2021-01-11:
If you open the "History Browser" you get an empty diagram (without data):

[image]

You have to press the "24" button (in the upper right corner), then the data is displayed:
[image]

Same thing, if you want to switch the date (no matter which button you use): nothing happens! Again, you first have to press the "time range button" in the upper right corner, to get the results.

robertrub, 2021-01-13:
It was the same in 2.7.0 too. 
And the "Button" text on button is not very explicite ;)

rICTx-T1D, 2021-01-23:
2.8.2 master is affected too.

MilosKozak, 2021-06-23:
resolved in dev

closed 2021-06-23 by MilosKozak