nightscout/AndroidAPS #219: [2.8.0.1_dev] wear: cannot disable vibration during bolus

guydavies, 2021-01-12:
Using 2.8.0.1_dev build fb1a4d40b-2021.01.11 I have tried to disable the vibration on each increment of bolus insulin. I've tried to use the settings on the watch and vibrate on bolus is off but the watch still vibrates every 0.2u. It appears to be possibly linked to the refresh of the screen with the cancel message. 

I would prefer a single vibration at the end of the bolus and maybe one at the start. If you're dosing 12u then all the buzzes become very annoying (60 of them). I'd prefer to be able to completely disable the vibration too. I have a vibration on my pump.

rICTx-T1D, 2021-01-12:
Do you bow the last version how it works correctly or is it a new feature request/ improvement?

guydavies, 2021-01-12:
This is still true for 2.8.1.1.

AdrianLxM, 2021-01-12:
The reason behind always vibrating was (4 years ago) that watches wouldn't show the notification as popup reliably if it was not together with a vibration.
We therefore used a non-noticable vibration pattern (vibrate one millisecond every second).

In https://github.com/nightscout/AndroidAPS/pull/229 I did the same but my tic-watch still vibrates a tiny bit. Can you check if that's ok for you now? (You might need to restart the watch after installing or switching off the setting if you had a notification already).

guydavies, 2021-01-13:
Testing it now.  I’ll let you know after lunch (in about an hour or so) if it’s working for me and if it is sufficiently light to not be annoying :)

Thanks for the quick efforts to find a workaround.  I hope something more complete can be done to fix the actual issue that the setting of vibrate on bolus to “off” doesn’t actually turn off the vibrations on bolus.

Rgds,

Guy

guydavies, 2021-01-13:
Hmm. I forgot to restart my TicWatch so it didn't seem any better at lunch.
I have now restarted the watch so let's see what happens at dinner time
this evening.

closed 2021-01-13 by MilosKozak

guydavies, 2021-01-13:
The update doesn't really improve anything for me with the TicWatch S2. I've tried restarting the watch and it still buzzes too much to be ignored.

guydavies, 2021-01-14:
Please reopen this. My watch is still buzzing for all boluses including SMB.  It's really annoying and a definite step back from previous implementations.

guydavies, 2021-01-14:
Never mind. I updated to the latest Dev and that does seem to work as expected. I see the status on the watch but I don't feel the vibrations. I am not sure why this worked this time. Last time I did an install using adb while this time I removed and installed on the watch and then ensured that the settings are right. Maybe that different approach is important.

rICTx-T1D, 2021-01-14:
I had the same problem and solved it myself. 
On Wear:
1. Open settings
2. Apps and Notifications 
3. Notifications 
4. All
5. Select AAPS
6. Switch off "AAPS Bolus progress "