nightscout/AndroidAPS #227: Since Update 2.8.1.1 history is empty

FreDiabetics, 2021-01-12:
Like the title says, since update to 2.8.1.1 the history  is totally empty. Doesn't matter how date I select nothings gets loaded. 

I'm using a different aaps version, but my looper-dude has noticed too and he's using original aaps version.

In 2.8.0 history was working

[image]

osodebailar, 2021-01-12:
Yes , same here with 2.8.1.1

rICTx-T1D, 2021-01-12:
Maybe the l8cal database is empty after update from previous version... Give some hours or days and check it again.

guydavies, 2021-01-12:
It's been like that for over 24 hours but with 2.8.0.1 Dev then with 2.8.1.1

guydavies, 2021-01-12:
Sorry. Wrong issue.

robertrub, 2021-01-13:
I'm still in 2.7.0 and my history _opens_ empty too. But, if I hit the time button upper right, the data fils in. Give it a try.
I'll see how 2.8.1.1 works this afternoon.
[image]

Alamo04, 2021-01-17:
Confirmed in Master 2.8.1.1 and also in Dev. 2.8.1.2

radu-neacsu, 2021-01-24:
I think I found another related issue. If you add the carbs from nightscout they are not downloaded into the AAPS treatments. 
 Seems related. AAPS 2.8.0 working but 2.8.2 and 2.81.1 stopped importing/downloading information from nightscout

robertrub, 2021-01-24:
@radu-neacsu No, what you describe is not a bug but a feature :)
There is an bug in NS that caused errors in AAPS (too much insulin delivery) when data was D/Led from NS to AAPS. For security reasons, until the NS bug is resolved, AAPS will only UPLOAD to NS , D/L is turned off.

radu-neacsu, 2021-01-24:
@robertrub I wasn't aware of that. I will have some issues with this "feature". My system was dependent on the download feature. 
I am using Diabetes-m to calculate my meals, then a [python script](https://github.com/radu-neacsu/dm2nsc/blob/master/sync.py) that syncs the diabetes-m cloud with nightscout, and the AAPS is aware of what I eat.
Thanks for your response.

robertrub, 2021-01-24:
@radu-neacsu You can turn D/L from NS _ON_ if you put engineering mode but you should be well aware of the risks and the way they may or may not affect you.

markvader, 2021-01-24:
using engineering mode is the only way currently to switch NS_UPLOAD_ONLY to false.

FreDiabetics, 2021-01-24:
PLEASE OPEN A NEW ISSUE THID HAD NOTHING TO DO WITH THE MENTIONED PROBLEMS! THANK YOU

szantos, 2021-01-26:
Same history issues here, even with the NSClient app on another phone. Also present on 2.8.2

AdrianLxM, 2021-02-09:
Duplicate of https://github.com/nightscout/AndroidAPS/issues/217
Please follow updates there.

closed 2021-02-09 by AdrianLxM